alibaba/AliOS-Things #1055: 安装aos-cube报错

EDGDK, 2019-10-15:
pip安装aos-cube报错，报错信息如下：
 ERROR: Command errored out with exit status 1:
     command: 'c:\users\74516\appdata\local\programs\python\python37-32\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\74516\\AppData\\Local\\Temp\\pip-install-1xfi1ezt\\aos-cube\\setup.py'"'"'; __file__='"'"'C:\\Users\\74516\\AppData\\Local\\Temp\\pip-install-1xfi1ezt\\aos-cube\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\74516\AppData\Local\Temp\pip-install-1xfi1ezt\aos-cube\pip-egg-info'
         cwd: C:\Users\74516\AppData\Local\Temp\pip-install-1xfi1ezt\aos-cube\
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\74516\AppData\Local\Temp\pip-install-1xfi1ezt\aos-cube\setup.py", line 4, in <module>
        from aos import __title__, __version__, __description__, __url__, __author__, __email__
      File "C:\Users\74516\AppData\Local\Temp\pip-install-1xfi1ezt\aos-cube\aos\__init__.py", line 17, in <module>
        sys.stderr.write(msg % (__version__, sys.version))
    TypeError: not all arguments converted during string formatting
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

**当前环境**
1. python3.7.4，setuptools（41.4.0)，pip（19.3）
2. win10

求各位大佬解答，按照官方案例搭开发环境出现的错误

xiaowenxia, 2019-10-16:
`aos-cube`目前还不支持python3。

EDGDK, 2019-10-17:
> `aos-cube`目前还不支持python3。

谢谢我去试试，之前我用python2.7他提醒我更新 不更新到最新不影响吗

xiaowenxia, 2019-10-21:
没有影响的。

AaronLiu92, 2020-02-18:
> 没有影响的。

你好，请问我今天安装aos-cube，有个spawn virtualenv enoent错误，是为啥？跟python2.7今年年初寿命已尽有关？谢谢

closed 2022-03-01 by skylarCai