alibaba/AliOS-Things #1058: 安装编译环境提示错误。Command: pip2 not exist, Please install pip2

iiiilii, 2019-10-18:
打开VS CODE提示Command: pip2 not exist, Please install pip2

环境变量已经设置。VS CODE终端敲命令 PIP  python均能识别

iiiilii, 2019-10-23:
上面报错解决，更新WIN10系统到最近即可。

但编译例程出现新报错

|=================================================================|
python.exe: can't open file './platform/mcu/moc108//padzero.py': [Errno 2] No such file or directory
make.exe[1]: *** [platform/mcu/moc108//gen_image_bin.mk:35: gen_standard_images] Error 2
make.exe: *** [build/Makefile:393: main_app] Error 2
[AliOS-Things] ERROR: "E:/AliOS-Things/AliOS-Things-rel_3.0.0/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/AliOS-Things/AliOS-Things-rel_3.0.0/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/AliOS-Things/AliOS-Things-rel_3.0.0/build -e -f build/Makefile" in "E:\AliOS-Things\AliOS-Things-rel_3.0.0"

xiaowenxia, 2019-10-23:
请更新AliOS Things 到最新的代码哈：

```
$ git pull
```

iiiilii, 2019-10-23:
> 请更新AliOS Things 到最新的代码哈：
> 
> ```
> $ git pull
> ```

OK，可以了

closed 2022-03-01 by skylarCai