alibaba/AliOS-Things #1061: https://github.com/alibaba/AliOS-Things.git 代码根本无法下载，下载到20%左右直接停了，你们服务器到底怎么搞得，这么难开发

tanglv, 2019-10-23:
**Describe the bug**

**environment**
1. branch(version): AliOS-Things
2. ubuntu16.04
3. example: 

**Expected behavior**

**Screenshots**
[image]

**Additional context**

xiaowenxia, 2019-10-23:
github在国内确实下载比较慢，除了从github下载，还有以下方式可以获取AliOS Things 源码：

* 1. 从gitee下载源码：
```
git clone git@gitee.com:mirrors/AliOS-Things.git
```
gitee主页：https://gitee.com/mirrors/AliOS-Things
* 2. 参考 #1054 > **如何快速git clone github上的源码** 中的方式快速clone github源码。

* 3. 在[AliOS Things组件在线裁剪工具](https://aliosthings.iot.aliyun.com/)中定制AliOS Things 源码。

panguoxian, 2019-11-30:
https://gitee.com/alios-things/AliOS-Things/tree/master

walker2048, 2020-03-16:
github现在是微软的了，不关人家阿里的事儿，访问慢很多原因

closed 2022-03-01 by skylarCai