alibaba/AliOS-Things #1062: 按照Quick Start.zh文档，搭建环境，执行aos make helloworld@linuxhost -c config && aos make报错

tanglv, 2019-10-23:
**Describe the bug**
按照Quick Start.zh文档，搭建环境，执行aos make helloworld@linuxhost -c config && aos make报错，log如下
执行命令
aos make helloworld@linuxhost -c config && aos make
[image]

换命令执行
aos make helloworld@linuxhost -c profile-linkkitapp_asr5501_defconfig  && aos make

[image]

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):ubuntu16.04
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

tanglv, 2019-10-23:
w10环境编译ok，但是linux环境不行

xiaowenxia, 2019-10-23:
`uname -a` 看看系统信息。

tanglv, 2019-10-23:
[image]

tanglv, 2019-10-23:
> `uname -a` 看看系统信息。

[image]

xiaowenxia, 2019-10-23:
你是windows上挂载的AliOS Things文件夹到ubuntu虚拟机么？

tanglv, 2019-10-24:
> 你是windows上挂载的AliOS Things文件夹到ubuntu虚拟机么？
没有挂在文件系统，是直接在/home目录下面的家目录，环境都按照文档安装好了，编译报错
[image]

tanglv, 2019-10-25:
> 你是windows上挂载的AliOS Things文件夹到ubuntu虚拟机么？

哈喽，还有人吗？帮忙解决一下，ubuntu开发环境也是主流啊

xiaowenxia, 2019-10-25:
@tanglv 你的源码下载的对吗？是不是没有下载完全？ubuntu系统是比较常用的一个开发环境，一直是有测试验证的。

closed 2022-03-01 by skylarCai