alibaba/AliOS-Things #1063: 怎么用python编写aliosthing

sunhao-jz, 2019-10-24:
我现在想用python去编写aliosthings，有没有相关文档的介绍

xiaowenxia, 2019-10-24:
没懂你的意思，micropython？

sunhao-jz, 2019-10-24:
就是我用python 的语言去写程序，然后直接可以编译生成可编译的文件下载到mcu里

xiaowenxia, 2019-10-24:
AliOS Things支持 micropython：https://github.com/alibaba/AliOS-Things/tree/rel_3.0.0/3rdparty/experimental/micropython 。

不过micropython不会编译，是执行python脚本。

sunhao-jz, 2019-10-24:
那有没有用js编写的相关demo或者文档？

xiaowenxia, 2019-10-24:
有的，js相关的文档在这里：https://studio.iot.aliyun.com/device-doc#ebagtb.html
编译固件：

```
aos make jsengine_app@esp32devkitc -c config
aos make
```

编写js脚本用这个webIDE：https://linkdevelop.aliyun.com/deviceapp#/s

sunhao-jz, 2019-10-24:
TinyEngine 已支持硬件列表我看stm32f103的支持的OS是ucos 不支持alios是吗？可以自己移植吗？还是官方以后会移植？

xiaowenxia, 2019-10-24:
JSEngine目前不支持stm32f103。可以自己移植，不过自己移植会比较麻烦，目前正在计划出JSEngine的适配指南。

sunhao-jz, 2019-10-24:
这个适配指南主要是做哪方面的？知道如何连接JSEngine 和 底层硬件的吗？

xiaowenxia, 2019-10-24:
目前我们支持的开发板有限：`esp32devkitc`、`developerkit`、`stm32f429zi-nucleo`，出适配指南主要是可以让开发者自己适配JSEngine到自己的开发板上。

sunhao-jz, 2019-10-24:
了解了，那我刚才看js嵌入式的准备工作，怎么stm32固件烧录指南打不开？

sunhao-jz, 2019-10-24:
C:\Users\Sunhao\AppData\Roaming\npm\node_modules\be-cli\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
gyp ERR! find VS
gyp ERR! find VS msvs_version not set from command line or npm config
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS could not use PowerShell to find Visual Studio 2017 or newer
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS
gyp ERR! configure error
gyp ERR! stack Error: Could not find any Visual Studio installation to use
gyp ERR! stack     at VisualStudioFinder.fail (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:121:47)
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:74:16
gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:351:14)
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:70:14
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:372:16
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:302:5)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at maybeClose (internal/child_process.js:1028:16)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Sunhao\AppData\Roaming\npm\node_modules\be-cli\node_modules\serialport
gyp ERR! node -v v13.0.1
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Sunhao\AppData\Roaming\npm-cache\_logs\2019-10-24T06_26_57_165Z-debug.log
我检查了python 我是2.7.16版本  环境变量是D:\Program Files (x86)\STMicroelectronics\st_toolset\asm;C:\Python27\;C:\Python27\Scripts;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\WIDCOMM\Bluetooth Software\;C:\Program Files\WIDCOMM\Bluetooth Software\syswow64;D:\Program Files\TortoiseSVN\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Git\cmd;C:\Program Files (x86)\GNU Tools ARM Embedded\5.4 2016q3;C:\Soft\Keil_v5\ARM\ARMCC\bin;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;D:\Program Files\doxygen\bin;C:\Program Files\nodejs\   里面有python的，为什么还是出这个错误？

closed 2022-03-01 by skylarCai