alibaba/AliOS-Things #1076: aos make helloworld@hostlinux -c config报错

zt19901990, 2019-11-20:
**Describe the bug**
执行aos make helloworld@hostlinux -c config报错
AliOS/AliOS-Things$ aos make helloworld@hostlinux -c config
aos-cube version: 0.3.9
Creating .defconfig ...
Creating .config ...
#
# configuration written to .config
#
Checking config output: .config ...
[ERROR]: App name Mismatched! (helloworld -> athostapp)
- Please config items for it are correct!

build/build_rules/aos_kconfig.mk:139: recipe for target '.config' failed
make: *** [.config] Error 1
[AliOS-Things] ERROR: "/mnt/e/AliOS/AliOS-Things/build/cmd/linux64/make" returned error code 2.
[AliOS-Things] ERROR: Command "/mnt/e/AliOS/AliOS-Things/build/cmd/linux64/make HOST_OS=Linux64 TOOLS_ROOT=/mnt/e/AliOS/AliOS-Things/build -e -f build/Makefile helloworld@hostlinux.config" in "/mnt/e/AliOS/AliOS-Things"

aos make menuconfig可以正常配置，编译。

**environment**
1. branch(version): rel_3.0.0
2. Desktop OS(windows linux or macOS): win10 linux子系统WSL
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

andy2012zwj, 2019-11-26:
用这个命令报错吗，aos make helloworld@linuxhost -c config

zt19901990, 2019-11-28:
同样的错误
Checking config output: .config ...
[ERROR]: App name Mismatched! (helloworld -> athostapp)

rolandma, 2019-12-04:
Same here. aos make helloworld@developerkit -c config same error.

closed 2022-03-01 by skylarCai