alibaba/AliOS-Things #1081: ubuntu 下 aos make helloworld@linuxhost -c config 出现 recipe for target '.config' failed 编译不过

panguoxian, 2019-11-30:
ubuntu 
aos make helloworld@linuxhost -c config
出现很多  ignoring unsupported character 

然后就 build/build_rules/aos_kconfig.mk:139: recipe for target '.config' failed

如何解决？是不是我环境有什么东西没有安装？

'est/yunit/Config.in:1:warning: ignoring unsupported character '
'est/yunit/Config.in:2:warning: ignoring unsupported character '
'est/yunit/Config.in:3:warning: ignoring unsupported character '
'est/yunit/Config.in:4:warning: ignoring unsupported character '
'est/yunit/Config.in:4:warning: ignoring unsupported character '
'est/Config.in:6:warning: ignoring unsupported character '
build/build_rules/aos_kconfig.mk:139: recipe for target '.config' failed
make: *** [.config] Error 1
[AliOS-Things] ERROR: "/media/share/AliOS-Things/build/cmd/linux64/make" returned error code 2.
[AliOS-Things] ERROR: Command "/media/share/AliOS-Things/build/cmd/linux64/make HOST_OS=Linux64 TOOLS_ROOT=/media/share/AliOS-Things/build -e -f build/Makefile helloworld@exp8266.config" in "/media/share/AliOS-Things"

rolandma, 2019-12-04:
When using git, you need to set use LFS, because the git automatically changed the line end to CRLF for windows.

panguoxian, 2019-12-09:
> When using git, you need to set use LFS, because the git automatically changed the line end to CRLF for windows.
thank

Hanxiaomo9527, 2020-03-02:
我也出现这个问题了，请问如何处理呢？

rolandma, 2020-03-02:
> 我也出现这个问题了，请问如何处理呢？

Depends on which git are you using, delete the repository, clone again with the option "Use Git LFS" selected. Use TortoiseGit, you can see the check box when trying to clone. 
Basically you are using Windows and the git default setting will convert all the line ends into CRLF, which causes your problem.

closed 2022-03-01 by skylarCai