alibaba/AliOS-Things #1085: 新建立工程后，编译报错No such file no entry point!!

panguoxian, 2019-12-01:
编译SDK工程的example没有问题。

但是新建立工程保存到别的地方后编译出现如下问题。

> Executing task: aos make <

aos-cube version: 0.3.9
Check if required tools for esp8266 exist
ECHO is off.
Build Configuration:
====================
App:   blinkapp
Board: esp8266
====================
ECHO is off.
Build AOS Now
TOOLCHAIN_PATH=
INFO: Could not find files for the given pattern(s).
Making blinkapp@esp8266.bin

                        AOS MEMORY MAP
|=================================================================|
| MODULE                                   | ROM       | RAM      |
|=================================================================|
| arch_xtensa_lx106                        | 4016      | 2784     |
| blinkapp                                 | 14        | 0        |
| board_esp8266                            | 1040      | 0        |
| debug                                    | 4100      | 16       |
| kernel_init                              | 108       | 0        |
| kv                                       | 2244      | 24       |
| libcirom                                 | 27513     | 1200     |
| libcrypto                                | 14506     | 0        |
| libgcc                                   | 2167      | 0        |
| libhal                                   | 37        | 0        |
| liblwip                                  | 31201     | 1688     |
| libmain                                  | 30411     | 708      |
| libnet80211                              | 42675     | 3213     |
| libphy                                   | 48158     | 461      |
| libpp                                    | 43996     | 18013    |
| libwpa                                   | 17240     | 626      |
| mcu_esp8266                              | 8413      | 587      |
| netmgr                                   | 80        | 8        |
| newlib_stub                              | 363       | 0        |
| osal_aos                                 | 721       | 0        |
| rhino                                    | 12191     | 3236     |
| ulog                                     | 589       | 7        |
| vfs                                      | 1340      | 1105     |
| yloop                                    | 1992      | 32       |
| *fill*                                   | 1947      | 128      |
|=================================================================|
| TOTAL (bytes)                            | 297062    | 33836    |
|=================================================================|
"Generate standard flash images C:/Users/Administrator/aosApp/blinkapp/out/blinkapp@esp8266/binary/blinkapp@esp8266.elf"
E:/VM/share/AliOS-Things/build/compiler/gcc-xtensa-lx106/Win32/bin/xtensa-lx106-elf-nm.exe: 'blinkapp@esp8266.elf': No such file
no entry point!!
E:/VM/share/AliOS-Things/build/cmd/win32/cp.exe: cannot stat `C:/Users/Administrator/aosApp/blinkapp/out/blinkapp@esp8266/binary/eagle.app.flash.bin': No such file or directory
make.exe[1]: *** [E:/VM/share/AliOS-Things/platform/mcu/esp8266//gen_image_bin.mk:24: gen_standard_images] Error 1
make.exe: *** [E:/VM/share/AliOS-Things/build/Makefile:393: main_app] Error 2
[AliOS-Things] ERROR: "E:/VM/share/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/VM/share/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/VM/share/AliOS-Things/build -e -f E:/VM/share/AliOS-Things/build/Makefile SOURCE_ROOT=E:/VM/share/AliOS-Things APPDIR=C:\Users\Administrator\aosApp\blinkapp BUILD_DIR=C:/Users/Administrator/aosApp/blinkapp/out" in "E:\VM\share\AliOS-Things"
---
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

andy2012zwj, 2019-12-06:
这个是aos哪个版本上出现的问题

panguoxian, 2019-12-20:
> 这个是aos哪个版本上出现的问题

3.0 版本的

closed 2022-03-01 by skylarCai