alibaba/AliOS-Things #1087: 移植到芯片stm32f103c8, 用alios编译helloword的时候，报make.exe: *** [build/Makefile:393: main_app] Error 2.

unitsoftrover, 2019-12-06:
该程序移植后通过KEIL建立工程，编译已经能够正常运行。
在alios中编译经常出错，怀疑是python不支持unicode导致，转换为纯ANSI，把含有中文的乱码去掉后，有时可以编译通过，有时编译无法通过。

App:   helloworld
Board: stm32f103c8-nucleo
====================
ECHO 处于关闭状态。
Build AOS Now
TOOLCHAIN_PATH=
Making helloworld@stm32f103c8-nucleo.bin

                        AOS MEMORY MAP
|Traceback (most recent call last):
  File ".//build/scripts/map_parse_gcc.py", line 105, in <module>
    main()
  File ".//build/scripts/map_parse_gcc.py", line 102, in main
    get_mem_info(sys.argv[1])
  File ".//build/scripts/map_parse_gcc.py", line 92, in get_mem_info
    print('|=================================================================|')
IOError: [Errno 0] Error
make.exe[1]: *** [build/build_rules/aos_target_build.mk:350: display_map_summary] Error 1
make.exe: *** [build/Makefile:393: main_app] Error 2
[AliOS-Things] ERROR: "G:/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "G:/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=G:/AliOS-Things/build -e -f build/Makefile" in "G:\AliOS-Things"
---
The terminal process terminated with exit code: 1

unitsoftrover, 2019-12-06:
有时候报这样的错误，看起来编译部分是成功了。

Making helloworld@stm32f103c8-nucleo.hex
| board_stm32f103c8-nucleo                 | 1611      | 528      |
| debug                                    | 3206      | 16       |
| helloworld                               | 4         | 0        |
| kernel_init                              | 2459      | 0        |
| libc_nano                                | 10400     | 464      |
| libgcc                                   | 3124      | 0        |
| mcu_stm32f1xx                            | 6163      | 373      |
| newlib_stub                              | 276       | 0        |
| osal_aos                                 | 492       | 0        |
| rhino                                    | 11123     | 3397     |
| ulog                                     | 96        | 7        |
| vfs                                      | 402       | 1105     |
| *fill*                                   | 48        | 11710    |
|=================================================================|
| TOTAL (bytes)                            | 40418     | 17600    |
|=================================================================|
Generate Raw OTA image: out/helloworld@stm32f103c8-nucleo/binary/helloworld@stm32f103c8-nucleo_ota.bin ...
Generate Compressed OTA image: out/helloworld@stm32f103c8-nucleo/binary/helloworld@stm32f103c8-nucleo_ota.bin.xz ...
41292
8033c0d77b9ea5b3e4f24a41ff54af36
25512
0bdb7b838f321e873ff31147cdf7849a
make.exe[1]: write error: stdout
make.exe: *** [build/Makefile:393: main_app] Error 1
[AliOS-Things] ERROR: "G:/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "G:/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=G:/AliOS-Things/build -e -f build/Makefile" in "G:\AliOS-Things"

unitsoftrover, 2019-12-06:
看来AliOS-studio远没有KEIL MDK好用。AliOS-Studio生成的KEIL工程也没法编译，花了好大功夫把AliOS工程转换为KEIL工程，里面的汇编启动程序坑爹，做了很大修改才能在KEIL下运行。

xiaowenxia, 2019-12-06:
@unitsoftrover alios things编译完后，会默认执行`//build/scripts/map_parse_gcc.py`，`map_parse_gcc.py`这个脚本是去解析gcc编译出来的map文件的，你在keil工程中，就没必要调用`map_parse_gcc.py`了。

unitsoftrover, 2019-12-07:
在keil中编译运行都正常的，我想在vscode中进行编译和调试，发现这个build/scripts/map_parse_gcc.py错误，怎么解决这个问题呢？

closed 2022-03-01 by skylarCai