alibaba/AliOS-Things #1099: esp8266 使用linkkitapp例程修改三元组失效

panguoxian, 2020-01-04:
esp8266  使用linkkitapp例程，修改了三元组，但是连接的时候还是使用修改前的。

默认是
#define PRODUCT_KEY      "a1FxISeKbq9"
#define PRODUCT_SECRET   "ThNbP5iNUQ1lQe2Q"
#define DEVICE_NAME      "alen-activate-test"
#define DEVICE_SECRET    "jcumDL5AJRgU7zRNcCcnHRiQmtii0vDn"

修改过以上参数后，编译烧录，连接时打印的还是以上默认的参数。

搞了一晚上不知哪里出的问题。。

andy2012zwj, 2020-01-05:
能确认下用的是AliOS Things的哪个版本吗

panguoxian, 2020-01-05:
> 能确认下用的是AliOS Things的哪个版本吗，如果是3.0版本，用这个命令设置（四个位置分别填入要设置的四元组）：devinfo set "PRODUCT_KEY" "PRODUCT_SECRET" "DEVICE_NAME" "DEVICE_SECRET"

不是应该在linkkit_example_solo.c里面修改吗？

panguoxian, 2020-01-05:
> 能确认下用的是AliOS Things的哪个版本吗，如果是3.0版本，用这个命令设置（四个位置分别填入要设置的四元组）：devinfo set "PRODUCT_KEY" "PRODUCT_SECRET" "DEVICE_NAME" "DEVICE_SECRET"

注释了set_iotx_info()里的HAL_GetDeviceName(_device_name);就可以了，因为之前有烧录过默认的，保存了，修改了新的再烧录就不会修改了。

void set_iotx_info()
{
    char _device_name[IOTX_DEVICE_NAME_LEN + 1] = {0};
    // HAL_GetDeviceName(_device_name);
    if (strlen(_device_name) == 0) {
        HAL_SetProductKey(PRODUCT_KEY);
        HAL_SetProductSecret(PRODUCT_SECRET);
        HAL_SetDeviceName(DEVICE_NAME);
        HAL_SetDeviceSecret(DEVICE_SECRET);
    }
}

andy2012zwj, 2020-01-05:
默认都需要从kv里读取的，更换了四元组之后清理一下kv也可以

closed 2022-03-01 by skylarCai