alibaba/AliOS-Things #1105: JSEngine App build

CJayakumar, 2020-01-15:
Could you please provide the steps to build jsengine_app.

When we execute the below command,

aos make jsengine_app@developerkit -c config
aos make

Then we can see till below lines in the console,

> Executing task: aos make <

aos-cube version: 0.3.10
Check if required tools for developerkit exist
ECHO is off.
Build Configuration:
====================
App:   jsengine_app
Board: developerkit
====================
ECHO is off.
Applying changes made to: out/config/autoconf.h
app_version:app-1.0.0-20200115.212152
processing components: jsengine_app developerkit

After that we could not see any process..

Please help us to resolve the issue.

Thank you.

xiaowenxia, 2020-01-16:
@CJayakumar  Apologize...!  this build error has been fixed, check #1106 .

CJayakumar, 2020-01-16:
The build is successful, thanks

closed 2020-01-16 by CJayakumar