alibaba/AliOS-Things #1111: Please add AOS_SDK_PATH in environment variable, then Reboot

woshi520laoxie, 2020-01-20:
我在使用vscode配置的时候，报了这么一个问题
Please add AOS_SDK_PATH in environment variable, then Reboot
这个问题我没有百度到，而且前缀是AOD开头的，我猜测试添加一个交叉编译器的路径还是什么吧？
那么这个问题有解决方式吗？
还有你们的交叉有点问题
1、-安装 aos-cube的位置的图片看不了，网速很好都看不了。有点问题

xiaowenxia, 2020-01-21:
参考这篇文档哈：[AliOS Things 3.0 应用开发指南](https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-3.0-%E5%BA%94%E7%94%A8%E5%BC%80%E5%8F%91%E6%8C%87%E5%8D%97)。

woshi520laoxie, 2020-01-21:
你们有接入天猫精灵的教程吗？
我百度找别人的教程用的例子是:
1、例子:linkkitapp
2、开发板esp32
3、linkkitapp@developerkit
不过使用阿里云APP，我无法连接到wifi。
你们有自己的例子教程吗？

woshi520laoxie, 2020-01-21:
你们有接入天猫精灵的教程吗？
我百度找别人的教程用的例子是:
1、例子:linkkitapp
2、开发板esp32
3、linkkitapp@esp32devkitc       （是这个，ubuntu编译的esp32的开发环境）
不过使用阿里云APP，我无法连接到wifi。
你们有自己的例子教程吗？

Hipin, 2020-02-12:
以win7为例，在[计算机]->属性->系统变量，里面->新建->变量名：AOS_SDK_PATH，变量值：D:\ALIOS_IOT\GaussianPrince-AliOS-Things-master\AliOS-Things（这个是ALios things 的路径）

closed 2022-03-01 by skylarCai