alibaba/AliOS-Things #1113: error: 'for' loop initial declarations are only allowed in C99 mode

andyjefflin, 2020-01-21:
Hi, please check below issue when compile ["sensor cloud demo"] examples:
(https://github.com/alibaba/AliOS-Things/blob/d47236fb29aab4cb95af318ea4a4a605a97963dd/drivers/sensor/hal/sensor_hal.c#L265) 
drivers/sensor/hal/sensor_hal.c:265:5: error: 'for' loop initial declarations are only allowed in C99 mode
     for(int index = 0; (index < g_sensor_cnt) && (index < SENSOR_MAX_NUM); index++){
 }

andy2012zwj, 2020-01-21:
Hi, thanks for your advice.
Would you fix that by taking the variable definition out of the 'for' loop? Just like this:

    **int index;
    for(index = 0; (index < g_sensor_cnt) && (index < SENSOR_MAX_NUM); index++){**

Or you can open a ticket for us to handle that:
https://workorder.console.aliyun.com/console.htm#/ticket/add?productCode=iot&commonQuestionId=3576&isSmart=true&iatraceid=1579603151567-2921fdb6c5c9a2913b6dff&channel=selfservice

andyjefflin, 2020-01-22:
Thanks!
Please check the ticket "CTBA8BT".

closed 2022-03-01 by skylarCai