alibaba/AliOS-Things #1118: JSEngine LCD.rect函数的问题。

toopad, 2020-02-12:
我使用AliOS Things Developer Kit（上海诺行）编译并刷写JSEngine（Alios-Things_3.0.0）,LCD.open正常，LCD.show正常，但是LCD.rect花屏。是否是驱动代码的问题？测试代码如下：
var lcd_config = {};
lcd_config.height = 240;
lcd_config.width = 240;
var ret = LCD.open(lcd_config);
var lcd_fill = {};
lcd_fill.color = rgb565(0, 0, 0);
lcd_fill.sx = 0;
lcd_fill.sy = 0;
lcd_fill.ex = 239;
lcd_fill.ey = 239;
lcd_fill.fill = 1;
LCD.rect(lcd_fill);

toopad, 2020-02-22:
这个函数有问题吗？？？？？
static void lcd_draw_rect(int32_t x1, int32_t y1, int32_t x2, int32_t y2,
                          uint32_t color, uint8_t need_fill)
{
    int32_t i = 0;
    int32_t j = 0;
    if (need_fill) {
        for (i = y1; i <= y2; i++) {
            hal_lcd_draw_line(x1, i, color, x2 - x1 + 1);
        }
    } else {
        for (i = x1; i <= x2; ++i) {
            hal_lcd_draw_point(i, y1, color);
            hal_lcd_draw_point(i, y2, color);
        }
        for (i = y1; i <= y2; ++i) {
            hal_lcd_draw_point(x1, i, color);
            hal_lcd_draw_point(x2, i, color);
        }
    }
}

xiaowenxia, 2020-03-05:
您好，LCD的底层接口：[lcd_developerkit.c](https://github.com/alibaba/AliOS-Things/blob/rel_3.0.0/middleware/jsengine/src/port/hal/lcd_developerkit.c) 适配的有缺陷，可以参考下如下代码修复哈：

```diff
diff --git a/middleware/jsengine/src/port/hal/lcd_developerkit.c b/middleware/jsengine/src/port/hal/lcd_developerkit.c
index 6851e78a7..0f4db1ece 100644
--- a/middleware/jsengine/src/port/hal/lcd_developerkit.c
+++ b/middleware/jsengine/src/port/hal/lcd_developerkit.c
@@ -27,10 +27,17 @@ void hal_lcd_draw_point(uint16_t xpos, uint16_t ypos, uint16_t color)
     ST7789H2_WritePixel(xpos, ypos, color);
 }
 
+uint16_t rgb_buffer[240] = {0};
+
 void hal_lcd_draw_line(uint16_t xpos, uint16_t ypos, uint32_t rgb,
                        uint16_t length)
 {
-    ST7789H2_WriteLine(xpos, ypos, rgb, length);
+    int i = 0;
+    for(i = 0; i < length; i++ ){
+        rgb_buffer[i] = (uint16_t)rgb;
+    }
+
+    ST7789H2_WriteLine(xpos, ypos, rgb_buffer, length);
 }
 
 #endif

```

toopad, 2020-03-05:
谢谢，哈！

closed 2022-03-01 by skylarCai