alibaba/AliOS-Things #1123: aos_queue_recv不会阻塞

flyingfish23333, 2020-02-19:
我在测试udata_local_demo例程时发现当传感器读取出错时系统就会重启

环境
1. 我克隆的分支
commit 6c449089577cd99faa9b1f546cc5f421c3ff5a6b (HEAD -> master, origin/master, origin/HEAD)
Author: chenan.xxw <chenan.xxw@alibaba-inc.com>
Date:   Thu Jan 16 14:24:46 2020 +0800

    BugID: 19015096: update dingtalk qrcode in readme

commit a9819f2a3a0f9ebcc8c814f99d02b1498cd3bc03
Author: chenan.xxw <chenan.xxw@alibaba-inc.com>
Date:   Thu Jan 16 11:55:34 2020 +0800

    BugID: 24582105: fix jsengine_app compile error using aos-cube 0.3.10, fix jsengine:log is not defined

2. 主机是Windows，板子芯片是esp8266
3. 示例：
测试例程是udata_demo.udata_local_demo@esp8266

我在middleware/udata/abs_data_model/abs_data_model.c文件加了一些打印信息

[image]

然后我在读取传感器的接口直接让其返回失败

[image]

这个温湿度传感器的驱动代码是我自己添加的，是可以正确读取温湿度，偶尔会出现读取的数据校验失败，但是跟这个BUG没什么关系，这个读的接口我是让其直接返回-1。

然后就会出现aos_queue_recv不会被阻塞，一直返回8，最后系统重启。
这是打印信息

[image]

andy2012zwj, 2020-02-22:
板子上的I2C驱动有没有初始化，I2C配置问题的话会导致复位

flyingfish23333, 2020-02-24:
> 板子上的I2C驱动有没有初始化，I2C配置问题的话会导致复位

复位的原因是aos_queue_recv这里死循环了，因为假若我添加了休眠则不会复位，症结所在是aos_queue_recv在读取传感器数据失败后就无法阻塞了。

closed 2022-03-01 by skylarCai