alibaba/AliOS-Things #1124: AliOS-Things-rel_3.0.0\middleware\jsengine\src\engine\lite_engine\addons\lcd\module_lcd.c

toopad, 2020-02-22:
AliOS-Things-rel_3.0.0\middleware\jsengine\src\engine\lite_engine\addons\lcd\module_lcd.c

static void lcd_draw_rect(int32_t x1, int32_t y1, int32_t x2, int32_t y2,
                          uint32_t color, uint8_t need_fill)
{
    int32_t i = 0;
    int32_t j = 0;
    if (need_fill) {
        for (i = y1; i <= y2; i++) {
            hal_lcd_draw_line(x1, i, color, x2 - x1 + 1);
        }
    } else {
        for (i = x1; i <= x2; ++i) {
            hal_lcd_draw_point(i, y1, color);
            hal_lcd_draw_point(i, y2, color);
        }
        for (i = y1; i <= y2; ++i) {
            hal_lcd_draw_point(x1, i, color);
            hal_lcd_draw_point(x2, i, color);
        }
    }
}

toopad, 2020-02-22:
这个函数有问题吗？？？

andy2012zwj, 2020-02-22:
need_fill改为0，用hal_lcd_draw_point的方式，是否可以

closed 2020-02-23 by toopad

toopad, 2020-02-25:
hal_lcd_draw_point方法没有问题。是哪里的问题呢？

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用