alibaba/AliOS-Things #1135: IDE编译出现错误

Torchwoods, 2020-03-10:
环境：
windows 7 x64
alios-things rel_3.0.0
参考[AliOS Things 3.0 应用开发指南](https://developer.aliyun.com/article/720516?spm=a2c6h.12873639.0.0.16d7124aLD8Teh&groupCode=iot) 创建一个工程，然后编译，发现编译的时候，platform路径错误
things/example/myapp/out/myapp@mk3080/modules//platform/mcu/rtl8710bn/CheckSumUtils.o] Error 87

日志如下:
> Executing task: aos make <

aos-cube version: 0.3.11
Check if required tools for mk3080 exist
Parsing all components ...
ECHO 处于关闭状态。
Build Configuration:
====================
App:   myapp
Board: mk3080
====================
ECHO 处于关闭状态。
Creating E:/workspace/alios-things/example/myapp/out/config/autoconf.h ...
Making config file for first time
app_version:app-1.0.0-20200310.213617
processing components: myapp mk3080
*** All Components: myapp mk3080 mcu_rtl8710bn kernel_init arch_armv7m newlib_stub rhino lwip netmgr ulog vfs kv osal_aos yloop debug
Build AOS Now
TOOLCHAIN_PATH=
Compiling myapp
Compiling board_mk3080
Compiling mcu_rtl8710bn
process_begin: CreateProcess(d:\Program Files (x86)\GNU Tools ARM Embedded\5.4 2016q3\bin\arm-none-eabi-gcc.exe, arm-none-eabi-gcc.exe -c -MD -DMCU_FAMILY=\"mcu_rtl8710bn\" -DSYSINFO_PRODUCT_MODEL=\"ALI_AOS_MK3080\" -DSYSINFO_DEVICE_NAME=\"MK3080\" -L E:/workspace/alios-things/AliOS-Things/board/mk3080 -mcpu=cortex-m4 -mthumb -mthumb-interwork -mlittle-endian -DCONFIG_PLATFORM_8711B -DM3 -DHardFault_Handler=Ali_HardFault_Handler -w -ggdb -Os -Wall -Wfatal-errors -fsigned-char -ffunction-sections -fdata-sections -fno-common -std=gnu11 -DPLATFORM=\"mk3080\" -DSYSINFO_APP_VERSION=\"app-1.0.0-20200310.213617\" -include E:/workspace/alios-things/example/myapp/out/config/autoconf.h -Wall -Werror -Wno-unused-variable
-Wno-unused-parameter -Wno-implicit-function-declaration -Wno-type-limits -Wno-sign-compare -Wno-pointer-sign -Wno-uninitialized -Wno-return-type -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-value -Wno-strict-aliasing -Wall -Werror -Wno-unused-variable -Wno-unused-parameter -Wno-implicit-function-declaration -Wno-type-limits -Wno-sign-compare -Wno-pointer-sign -Wno-uninitialized -Wno-return-type -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-value -Wno-strict-aliasing -Wall -Werror -Wno-unused-variable -Wno-unused-parameter -Wno-implicit-function-declaration -Wno-type-limits -Wno-sign-compare -Wno-pointer-sign -Wno-uninitialized -Wno-return-type -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-value -Wno-strict-aliasing -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/. -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/../../../network/lwip/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/../../../network/lwip/include/lwip -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/arch -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/aos -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/peripherals -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/project/realtek_amebaz_va0_example/inc -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/os/os_dep/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/api/network/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/api -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/api/at_cmd -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/api/platform -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/api/wifi -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/api/wifi/rtw_wpa_supplicant/src -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/api/wifi/rtw_wowlan -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/api/wifi/rtw_wpa_supplicant/wpa_supplicant -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/modules -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/sdio/realtek/sdio_host/inc -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/inic/rtl8711b -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/usb_class/device -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/usb_class/device/class -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/wlan/realtek/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/wlan/realtek/src/osdep -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/wlan/realtek/src/hci -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/wlan/realtek/src/hal -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/wlan/realtek/src/hal/rtl8711b -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/wlan/realtek/src/hal/OUTSRC -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/drivers/wlan/realtek/wlan_ram_map/rom -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/network -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/app/monitor/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/cmsis -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/cmsis/device -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/fwlib -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/fwlib/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/fwlib/ram_lib/crypto -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/fwlib/rom_lib -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/swlib/os_dep/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/swlib/std_lib/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/swlib/std_lib/libc/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/swlib/std_lib/libc/rom/string -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/swlib/std_lib/libgcc/rtl8195a/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/swlib/rtl_lib -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/misc -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/misc/os -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/mbed/api -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/mbed/hal -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/mbed/hal_ext -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/mbed/targets/cmsis/rtl8711b -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/common/mbed/targets/hal/rtl8711b -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/./ -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/../include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/os/os_dep/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/app/monitor/include -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/cmsis -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/cmsis/device -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/sdk/component/soc/realtek/8711b/fwlib -IE:/workspace/alios-things/AliOS-Things/platform/mcu/rtl87 [... cut ...] HAL -DAOS_LOOP -DAOS_COMP_DEBUG -D__FILENAME__=\"CheckSumUtils.c\" -o E:/workspace/alios-things/example/myapp/out/myapp@mk3080/modules//platform/mcu/rtl8710bn/CheckSumUtils.o E:/workspace/alios-things/AliOS-Things/platform/mcu/rtl8710bn/CheckSumUtils.c, ...) failed.
make (e=87): 参数错误。
make.exe[1]: *** [E:/workspace/alios-things/AliOS-Things/build/build_rules/aos_target_build.mk:233: E:/workspace/alios-things/example/myapp/out/myapp@mk3080/modules//platform/mcu/rtl8710bn/CheckSumUtils.o] Error 87

andy2012zwj, 2020-03-17:
你好，这个路径名称太长，从而编译参数太长，导致错误，可以参考下面方式解决
1. 可以把OS代码从“E:/workspace/alios-things/AliOS-Things”改到“E:/aos”
2. 记得把AOS_SDK_PATH也同步修改下；
3. 再重新创建工程，编译；

Torchwoods, 2020-03-18:
> 你好，这个路径名称太长，从而编译参数太长，导致错误，可以参考下面方式解决
> 
> 1. 可以把OS代码从“E:/workspace/alios-things/AliOS-Things”改到“E:/aos”
> 2. 记得把AOS_SDK_PATH也同步修改下；
> 3. 再重新创建工程，编译；

ths

closed 2022-03-01 by skylarCai