alibaba/AliOS-Things #1136: 编译报错

includexiaozhongtao, 2020-03-13:
检测到 #include 错误。请更新 includePath。已为此翻译单元(E:\src\test\developerkit_sensor\developerkit_sensor\app_main.c)禁用波形曲线。无法打开 源 文件 "aos/kernel.h" (dependency of "E:\src\test\developerkit_sensor\developerkit_sensor\app_main.c")无法打开 源 文件 "stdio.h"无法打开 源 文件 "fcntl.h"无法打开 源 文件 "aos/kernel.h"无法打开 源 文件 "aos/cli.h"无法打开 源 文件 "k_api.h"无法打开 源 文件 "sensor/sensor.h"无法打开 源 文件 "linkkit/dev_model_api.h"无法打开 源 文件 "linkkit/infra/infra_compat.h"无法打开 源 文件 "linkkit/infra/infra_config.h"无法打开 源 文件 "linkkit/infra/infra_defs.h"无法打开 源 文件 "linkkit/infra/infra_types.h"无法打开 源 文件 "linkkit/wrappers/wrappers.h"无法打开 源 文件 "cJSON.h"检测到 #include 错误。请更新 includePath。已为此翻译单元(E:\src\test\developerkit_sensor\developerkit_sensor\linkkit_client.c)禁用波形曲线。无法打开 源 文件 "aos/kernel.h" (dependency of "E:\src\test\developerkit_sensor\developerkit_sensor\linkkit_client.c")无法打开 源 文件 "stdarg.h"无法打开 源 文件 "stdint.h"无法打开 源 文件 "stdio.h"无法打开 源 文件 "stdlib.h"无法打开 源 文件 "string.h"无法打开 源 文件 "aos/cli.h"无法打开 源 文件 "aos/kernel.h"无法打开 源 文件 "aos/yloop.h"无法打开 源 文件 "uagent/uagent.h"无法打开 源 文件 "ulog/ulog.h"无法打开 源 文件 "netmgr.h"无法打开 源 文件 "linkkit/dev_model_api.h"无法打开 源 文件 "linkkit/infra/infra_compat.h"无法打开 源 文件 "linkkit/infra/infra_config.h"无法打开 源 文件 "linkkit/infra/infra_defs.h"无法打开 源 文件 "linkkit/infra/infra_types.h"无法打开 源 文件 "linkkit/wrappers/wrappers.h"

andy2012zwj, 2020-03-13:
你好，可以把编译环境，执行的命令描述一下吗

includexiaozhongtao, 2020-03-13:
编译讲座源程序，点击左下角对号编译报的错误

includexiaozhongtao, 2020-03-13:
“developerkit_sensor”源程序

includexiaozhongtao, 2020-03-13:
将讲座列程打开，点击左下角对号编译，出现的错误

andy2012zwj, 2020-03-13:
> 编译讲座源程序，

讲座例程的链接可以发一下吗

includexiaozhongtao, 2020-03-13:
https://developer.aliyun.com/article/720812?spm=a2c6h.12873639.0.0.3bdd6ee3SNbrHW
这个链接提供的一个源码测试链接，直接下载编译后会报错误。

andy2012zwj, 2020-03-16:
你好，我这边试了下，编译没有问题，请确认以下两点：
1. 上面文档里的AliOS Things 3.0源码路径是否已经设置好；
2. AOS_SDK_PATH环境变量是否已经添加，https://developer.aliyun.com/article/720516?spm=a2c6h.13262185.0.0.70d24a00irxmWW#

includexiaozhongtao, 2020-03-16:
AOS_SDK_PATH已添加成功，AliOS Things 3.0源码路径已设置好

includexiaozhongtao, 2020-03-16:
无法打开 源 文件 "stdio.h"
无法打开 源 文件 "fcntl.h"
无法打开 源 文件 "aos/kernel.h"
无法打开 源 文件 "aos/cli.h"
无法打开 源 文件 "k_api.h"
无法打开 源 文件 "sensor/sensor.h"
无法打开 源 文件 "linkkit/dev_model_api.h"
无法打开 源 文件 "linkkit/infra/infra_compat.h"
无法打开 源 文件 "linkkit/infra/infra_config.h"
无法打开 源 文件 "linkkit/infra/infra_defs.h"
无法打开 源 文件 "linkkit/infra/infra_types.h"
无法打开 源 文件 "linkkit/wrappers/wrappers.h"
检测到 #include 错误。请更新 includePath。已为此翻译单元(E:\src\test\developerkit_sensor\developerkit_sensor\app_main.c)禁用波形曲线。
无法打开 源 文件 "aos/kernel.h" (dependency of "linkkit_client.h")

andy2012zwj, 2020-03-17:
编译环境是用的哪个，windoes平台上的vscode吗

includexiaozhongtao, 2020-03-17:
是的win10 Visual Studio Code

closed 2022-03-01 by skylarCai