alibaba/AliOS-Things #1141: ESP32 IIC驱动会导致死循环

panjian-suzhou, 2020-03-16:
**Describe the bug**
ESP32芯片SDK IIC驱动
static int32_t i2c_read_bytes(i2c_dev_t * handle,uint16_t addr,int8_t add_width,uint8_t * buff,uint32_t len,int8_t need_stop)
函数内部有个while(is_run);这个循环有一定的机率死在其中，只要一直读不到停止位，程序就死在这个循环里面了。

**environment**
1. branch(version): alios things 3.0.0
2. Desktop OS(windows linux or macOS):linux
3. example: 
我创建了一个时钟读取任务，每一秒通过IIC总线读取一次RX8025T，发现这个任务不定时的会“不运行”了，debug后，发现问题出在static int32_t i2c_read_bytes这个函数中。这个驱动应该还有其他问题，因为我的时钟有一定的机率读到异常数值，这个问题我还没解决。
**Expected behavior**

**Screenshots**

**Additional context**

andy2012zwj, 2020-03-16:
你好，这个问题可以提个工单吗：https://selfservice.console.aliyun.com/ticket/category/iot/recommend/3576

panjian-suzhou, 2020-03-16:
> 你好，这个问题可以提个工单吗：https://selfservice.console.aliyun.com/ticket/category/iot/recommend/3576
 好的，我已经提交了

closed 2022-03-01 by skylarCai