alibaba/AliOS-Things #1144: Developer kit JS编译报错

EienSakura, 2020-03-22:
[image]
[image]
这个open未定义怎么回事，我看似乎sample里是这样的，还有这个board.json应该怎么设置，port从哪里获取？

xiaowenxia, 2020-03-23:
@EienSakura open未定义，是因为需要

EienSakura, 2020-03-23:
> @EienSakura open未定义，是因为需要在menuconfig中配置使能gpio功能：
> 
> 进入：`Middleware` > `JSEngine` > `JSEngine Addon Modules` > `Hardware Addon Modules`
> 
> [image]
> 
> `board.json`怎么配置可以参考这篇文档：https://studio.iot.aliyun.com/device-doc#booofy.html

[image]
运行之后报了这个是怎么解决

xiaowenxia, 2020-03-23:
@EienSakura `board.json`怎么写的呢？提示board.json解析失败。

EienSakura, 2020-03-23:
> @EienSakura `board.json`怎么写的呢？提示board.json解析失败。
[image]

[image]
[image]

EienSakura, 2020-03-23:
{
  "I2C": [
    {
      "id": "shtc1",
      "port": 2,
      "address_width": 7,
      "freq": 100000,
      "mode": 1,
      "dev_addr": 224
    }
  ]
}

console.log('start i2c test.....................');
setInterval(function () {

  var shtc1Handle = I2C.open('shtc1');
  var Reg = [0xEF, 0xC8];
  I2C.write(shtc1Handle, Reg);
  var retArray = I2C.read(shtc1Handle, 2);
  console.log('ID:' + retArray[1]);
  Reg[0] = 0x80;
  Reg[1] = 0x5D;
  I2C.write(shtc1Handle, Reg);
  Reg[0] = 0x7C;
  Reg[1] = 0xA2;
  I2C.write(shtc1Handle, Reg);
  var TempHumm = I2C.read(shtc1Handle, 6);
  console.log('read len:' + TempHumm.length);
  var i = 0;
  for (i = 0; i < TempHumm.length; i++) {
    console.log('rawData[' + i + ']:' + TempHumm[i]);
  }
  I2C.close(shtc1Handle);
}, 2000);
console.log('end i2c test........................');

EienSakura, 2020-03-24:
@xiaowenxia 拜托大佬指导一下，我这边是想做一个往云平台上上传温湿度数据的项目，看wiki那边的例子又一直报错

closed 2022-03-01 by skylarCai