alibaba/AliOS-Things #1145: 移植stemwin的GUI报错

sunhao-jz, 2020-03-23:
_Min_Heap_Size = 0x400 ;	/* required amount of heap  */
_Min_Stack_Size = 0x1000 ;	/* required amount of stack */   这两个是设置什么？  当我将heap_Size s设置大之后，报错._user_heap_stack' will not fit in region `RAM，而我将这个设小反而不溢出了，为什么？

andy2012zwj, 2020-03-23:
你好，这是在哪个开发板上移植的，另外AliOS-Things的版本是多少

sunhao-jz, 2020-03-23:
在stm32f103vc 上移植，版本是3.0

andy2012zwj, 2020-03-23:
上面两个字段的含义，是系统堆和栈的大小，
ram大小为20K（如下所示），确认下_Min_Heap_Size增加后是否导致_user_heap_stack超过了ram大小：
RAM (xrw)      : ORIGIN = 0x20000000, LENGTH = 20K

sunhao-jz, 2020-03-23:
STM32f103vc的ram是48k的，而我设置的MIN_Heap_Size 设置0x400没事，设置成0x2800这是10k  按道理说远远没到48K  但是还是显示溢出了

andy2012zwj, 2020-03-23:
除了_user_heap_stack之外，data段和bss段也在ram空间里，可以确认下这部分的大小

sunhao-jz, 2020-03-23:
好，谢谢，还有一个问题c:/program files (x86)/gnu tools arm embedded/5.4 2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7-m\libm.a(lib_a-w_sqrt.o): In function `sqrt':
w_sqrt.c:(.text.sqrt+0x7e): undefined reference to `__errno'
w_sqrt.c:(.text.sqrt+0xa4): undefined reference to `__errno'
这个问题怎么解决啊？

sunhao-jz, 2020-03-23:
Compiling vfs
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/AliosGui.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/board_stm32f103vc-nucleo.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/mcu_stm32f1xx.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/osal_aos.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/kernel_init.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/arch_armv7m.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/newlib_stub.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/rhino.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/debug.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/ulog.a
Making C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/vfs.a
Making AliosGui@stm32f103vc-nucleo.elf
C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/libraries/AliosGui.a(GUIDEMO.o): In function `_Main':
C:/Users/Sunhao/Desktop/aliosGui/AliosGui/GuiApp/GUIDEMO.c:426: undefined reference to `GUIDEMO_Intro'
c:/program files (x86)/gnu tools arm embedded/5.4 2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7-m\libm.a(lib_a-w_sqrt.o): In function `sqrt':
w_sqrt.c:(.text.sqrt+0x7e): undefined reference to `__errno'
w_sqrt.c:(.text.sqrt+0xa4): undefined reference to `__errno'
collect2.exe: error: ld returned 1 exit status
make.exe[1]: *** [C:/Users/Sunhao/Desktop/AliOS-Things/build/build_rules/aos_target_build.mk:294: C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out/AliosGui@stm32f103vc-nucleo/binary/AliosGui@stm32f103vc-nucleo.elf] Error 1
make.exe: *** [C:/Users/Sunhao/Desktop/AliOS-Things/build/Makefile:393: main_app] Error 2
[AliOS-Things] ERROR: "C:/Users/Sunhao/Desktop/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "C:/Users/Sunhao/Desktop/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=C:/Users/Sunhao/Desktop/AliOS-Things/build -e -f C:/Users/Sunhao/Desktop/AliOS-Things/build/Makefile SOURCE_ROOT=C:/Users/Sunhao/Desktop/AliOS-Things APPDIR=C:\Users\Sunhao\Desktop\aliosGui\AliosGui BUILD_DIR=C:/Users/Sunhao/Desktop/aliosGui/AliosGui/out"
in "C:\Users\Sunhao\Desktop\AliOS-Things"
---

sunhao-jz, 2020-03-23:
上面的是详细的编译结果

closed 2020-03-23 by sunhao-jz