alibaba/AliOS-Things #1146: be-cli的编译问题

EienSakura, 2020-03-23:
[image]
这边这样子怎么解决，reboot也无法解决

andy2012zwj, 2020-03-23:
软件环境和编译命令可以描述一下吗

EienSakura, 2020-03-23:
编译命令：be -p bone
环境：windows
就是按着这个这个http://aliosthings.io/#/zh-cn/boneengine_firstjs?id=1-%e5%87%86%e5%a4%87%e5%b7%a5%e4%bd%9c做的

EienSakura, 2020-03-23:
开发板是developer kit

EienSakura, 2020-03-23:
> 软件环境和编译命令可以描述一下吗

编译命令：be -p bone
环境：windows
就是按着这个这个http://aliosthings.io/#/zh-cn/boneengine_firstjs?id=1-%e5%87%86%e5%a4%87%e5%b7%a5%e4%bd%9c做的

xiaowenxia, 2020-03-23:
@EienSakura be-cli工具使用，参考这个文档哈：[be-cli quickstart](https://github.com/alibaba/AliOS-Things/wiki/be-cli-quickstart)。

EienSakura, 2020-03-23:
> @EienSakura be-cli工具使用，参考这个文档哈：[be-cli quickstart](https://github.com/alibaba/AliOS-Things/wiki/be-cli-quickstart)。

[image]
还是有问题

closed 2022-03-01 by skylarCai