alibaba/AliOS-Things #1148: undefined reference to `__errno'  错误

sunhao-jz, 2020-03-23:
In function `sqrt':
w_sqrt.c:(.text.sqrt+0x7e): undefined reference to `__errno'
w_sqrt.c:(.text.sqrt+0xa4): undefined reference to `__errno'  这个错误怎么解决啊？

andy2012zwj, 2020-03-23:
我在helloworld里增加sqrt函数的调用，没出现这个错误，能确认下makefile是否有修改

sunhao-jz, 2020-03-23:
NAME := board_stm32f103vc-nucleo

$(NAME)_MBINS_TYPE := kernel
$(NAME)_VERSION    := 1.0.0
$(NAME)_SUMMARY    := configuration for board stm32f103vc-nucleo
MODULE             := 1062
HOST_ARCH          := Cortex-M3
HOST_MCU_FAMILY    := mcu_stm32f1xx
SUPPORT_MBINS      := no
HOST_MCU_NAME      := STM32F103VCT
ENABLE_VFP         := 0

$(NAME)_COMPONENTS += $(HOST_MCU_FAMILY) osal_aos kernel_init

$(NAME)_SOURCES += config/k_config.c \
                   config/partition_conf.c \
                   drivers/stm32f1xx_hal_msp.c \
                   drivers/system_stm32f1xx.c \
                   startup/board.c \
                   startup/startup.c
ywss_support ?= 0

GLOBAL_DEFINES += CONFIG_NO_TCPIP
GLOBAL_DEFINES += CONFIG_AOS_INIT_WELCOME
GLOBAL_DEFINES += STM32F103xE
#depends on sal module if select sal function via build option "AOS_NETWORK_SAL=y"
AOS_NETWORK_SAL	?= n

ifeq ($(COMPILER), armcc)
$(NAME)_SOURCES += startup/startup_stm32f103xc_keil.s
$(NAME)_LINK_FILES := startup/startup_stm32f103xc_keil.o
$(NAME)_LINK_FILES += drivers/stm32f1xx_hal_msp.o

GLOBAL_LDFLAGS += -L --scatter=board/stm32f103vc-nucleo/stm32f103vc-nucleo.sct
else ifeq ($(COMPILER), iar)
$(NAME)_SOURCES += startup/startup_stm32f103xc_iar.s
GLOBAL_LDFLAGS += --config board/stm32f103vc-nucleo/stm32f103xc_flash.icf
else
#GLOBAL_CFLAGS += -DUSE_DIRECT_UART_PUTC
$(NAME)_SOURCES += startup/startup_stm32f103xc.s
GLOBAL_LDFLAGS += -T board/stm32f103vc-nucleo/STM32F103VCTx_FLASH.ld
endif

GLOBAL_INCLUDES += . \
                   config/ \
                   drivers/

#GLOBAL_CFLAGS += -DSTM32F103xC -DCENTRALIZE_MAPPING -DHAL_SRAM_MODULE_ENABLED
GLOBAL_CFLAGS += -DSTM32F103xC
CONFIG_SYSINFO_PRODUCT_MODEL := ALI_AOS_stm32f103vc-nucleo
CONFIG_SYSINFO_DEVICE_NAME   := NUCLEO-F103VC

GLOBAL_CFLAGS += -DSYSINFO_PRODUCT_MODEL=\"$(CONFIG_SYSINFO_PRODUCT_MODEL)\"
GLOBAL_CFLAGS += -DSYSINFO_DEVICE_NAME=\"$(CONFIG_SYSINFO_DEVICE_NAME)\"
GLOBAL_CFLAGS += -DSYSINFO_OS_VERSION=\"$(CONFIG_SYSINFO_OS_VERSION)\"
GLOBAL_CFLAGS += -DSYSINFO_ARCH=\"ARM_Cortex-M3\"
GLOBAL_CFLAGS += -DSYSINFO_MCU=\"$(HOST_MCU_NAME)\"

# Keil project support
$(NAME)_KEIL_VENDOR = STMicroelectronics
$(NAME)_KEIL_DEVICE = STM32F103VC

# Iar project support: OGChipSelectEditMenu
$(NAME)_IAR_OGCMENU = STM32F103VC ST STM32F103VC

sunhao-jz, 2020-03-23:
这是我的makefile 我只是移植的时候改为了vc系列的

andy2012zwj, 2020-03-23:
这里看了下，应该没有问题，w_sqrt.c，这个文件能看到吗，里面的头文件包含是否准确

sunhao-jz, 2020-03-23:
这个文件我搜索不到，这个文件在哪里？

andy2012zwj, 2020-03-23:
https://github.com/alibaba/AliOS-Things/blob/rel_3.0.0/platform/mcu/stm32f1xx/aos.mk, 在这个文件的129行添加一下 -lc， 看看是否可以

sunhao-jz, 2020-03-23:
GLOBAL_LDFLAGS += -mcpu=cortex-m3 \
                  -mlittle-endian \
                  -mthumb -mthumb-interwork \
                  --specs=nosys.specs \
                  -mfloat-abi=soft \
                  $(CLIB_LDFLAGS_NANO_FLOAT) -lc
endif  
加了但是还是不行
c:/program files (x86)/gnu tools arm embedded/5.4 2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7-m\libm.a(lib_a-w_sqrt.o): In function `sqrt':
w_sqrt.c:(.text.sqrt+0x7e): undefined reference to `__errno'
w_sqrt.c:(.text.sqrt+0xa4): undefined reference to `__errno'   还是这个错误
collect2.exe: error: ld returned 1 exit status

andy2012zwj, 2020-03-23:
试一下stm32f103rb-nucleo单板的编译，是否可以

sunhao-jz, 2020-03-23:
单板是可以的，就是我移植的这个GUI需要用到MATH我包含了这个头文件后就出现这个问题了

andy2012zwj, 2020-03-23:
在helloworld里面添加一下sqrt的调用，是否也有同样的问题

sunhao-jz, 2020-03-23:
我试一下

sunhao-jz, 2020-03-23:
我是了这个函数是可以用的

andy2012zwj, 2020-03-23:
链接没有问题，那可以参考这个把你的GUI代码移植一下

sunhao-jz, 2020-03-23:
我用我移植的stm32F103VC系列的helloword 试了下 这个函数也是编译没问题的，但是我移植了stemwin就不行了

sunhao-jz, 2020-03-23:
我不包含gui的代码就没事，包含了就出问题

sunhao-jz, 2020-03-23:
因为之前我用keil移植gui没有问题，是不是KEIL的math和gcc的math的文件不一样？

andy2012zwj, 2020-03-23:
keil下面用的是armcc的话，和gcc编译出来的文件不一样，math的静态库不一样；另外GUI用的是源码还是静态库。这两点你都排查一下

sunhao-jz, 2020-03-23:
GUI我用的静态库，math在keil里我也是直接用的keil自带的math，gcc的math静态库本身自带吗？

andy2012zwj, 2020-03-23:
gcc里面也有math的静态库，确保GUI的静态库也是用gcc编译的

sunhao-jz, 2020-03-24:
现在stemwin 的GUI库是兼容gcc和keil的，不做区分的

closed 2020-03-26 by sunhao-jz