alibaba/AliOS-Things #1153: aos 生成不了keil工程

aktoey, 2020-04-01:
错误代码：
build/Makefile:125: *** Build Target Deprecated!.  Stop.

[AliOS-Things] ERROR: "E:/IDM_Download/RTOS/AliOS-Things-3.0.0/AliOS-Things-3.0.0/build\cmd/win32/make.exe" returned error code 2.

[AliOS-Things] ERROR: Command "E:/IDM_Download/RTOS/AliOS-Things-3.0.0/AliOS-Things-3.0.0/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/IDM_Download/RTOS/AliOS-Things-3.0.0/AliOS-Things-3.0.0/build -e -f build/Makefile helloworld@developerkit IDE=keil" in "E:\IDM_Download\RTOS\AliOS-Things-3.0.0\AliOS-Things-3.0.0"

andy2012zwj, 2020-04-01:
生成keil工程用的是哪个命令

aktoey, 2020-04-01:
> 生成keil工程命令是用的下面的这个吗？aos
> make helloworld @ developerkit -c config
> aos make IDE = keil

是的，是这个命令

andy2012zwj, 2020-04-01:
3.0版本的话，用这个命令试一下
aos make helloworld@developerkit -c config
aos make export-keil

aktoey, 2020-04-01:
谢谢，了能生成工程文件了，keil应该是没问题了，现在我用iar生成的工程编译不了，错误如下：
           --c99 (at line 1 of "E:\IDM_Download\RTOS\AliOS-Things-3.0.0\AliOS-Things-3.0.0\projects\IAR\helloworld@developerkit\iar_project\opts\rhino.c_opts") 
            --apcs=/hardfp (at line 1 of "E:\IDM_Download\RTOS\AliOS-Things-3.0.0\AliOS-Things-3.0.0\projects\IAR\helloworld@developerkit\iar_project\opts\rhino.c_opts") 
            -g (at line 1 of "E:\IDM_Download\RTOS\AliOS-Things-3.0.0\AliOS-Things-3.0.0\projects\IAR\helloworld@developerkit\iar_project\opts\rhino.c_opts")

aktoey, 2020-04-01:
还想问下，生成的keil和iar工程需要怎么配置才能正常编译，谢谢了