alibaba/AliOS-Things #1154: 用keil5和IAR8编译都出现错误

aktoey, 2020-04-02:
keil5 错误代码如下：
Fatal error: C3900U: Unrecognized option '-all'.
Warning: C3910W: Old syntax, please use '-E'.
Fatal error: C3900U: Unrecognized option '-rror'.
Fatal error: C3900U: Unrecognized option '-all'.
Warning: C3910W: Old syntax, please use '-E'.
Fatal error: C3900U: Unrecognized option '-rror'.
compiling ulog_utility.c...
Fatal error: C3900U: Unrecognized option '-all'.
Warning: C3910W: Old syntax, please use '-E'.
Fatal error: C3900U: Unrecognized option '-rror'.
Fatal error: C3900U: Unrecognized option '-all'.
Warning: C3910W: Old syntax, please use '-E'.
Fatal error: C3900U: Unrecognized option '-rror'.
compiling ulog_init.c...
Fatal error: C3900U: Unrecognized option '-all'.
Warning: C3910W: Old syntax, please use '-E'.
Fatal error: C3900U: Unrecognized option '-rror'.
Fatal error: C3900U: Unrecognized option '-all'.
Warning: C3910W: Old syntax, please use '-E'.
Fatal error: C3900U: Unrecognized option '-rror'.
".\Objects\helloworld@developerkit.axf" - 12 Error(s), 6 Warning(s).

IAR8 错误代码如下：
[Su002]: Unexpected command line arguments: 
            --c99 (at line 1 of "E:\IDM_Download\RTOS\AliOS-Things-3.0.0\AliOS-Things-3.0.0\projects\IAR\helloworld@developerkit\iar_project\opts\netmgr.c_opts") 
            --apcs=/hardfp (at line 1 of "E:\IDM_Download\RTOS\AliOS-Things-3.0.0\AliOS-Things-3.0.0\projects\IAR\helloworld@developerkit\iar_project\opts\netmgr.c_opts") 
            -g (at line 1 of "E:\IDM_Download\RTOS\AliOS-Things-3.0.0\AliOS-Things-3.0.0\projects\IAR\helloworld@developerkit\iar_project\opts\netmgr.c_opts") 
Error while running C/C++ Compiler

aktoey, 2020-04-02:
我觉得应该是工程配置的问题，还请提供正确的工程配置方法

andy2012zwj, 2020-04-03:
我试了下keil下面的，3.0版本编译出来的，没有问题，能确认下keil下编译器用的是哪个

aktoey, 2020-04-07:
用的是5.27 这个版本的，我觉得应该是配置的问题，但是我不知道是哪些配置出问题了

andy2012zwj, 2020-04-07:
3.0是指OS的版本，keil用的是5.23，生成keil工程用的是以下命令：
aos make helloworld@developerkit -c config
aos make export-keil
你这边看起来像是编译配置有问题；

aktoey, 2020-04-07:
我是用这些命令输出的工程文件，但是我不知道为什么会出现以上的错误提示

aktoey, 2020-04-07:
主要是Ulog那个文件夹的文件编译出现问题的

aktoey, 2020-04-07:
你可以把你生成的文件给我，试试吗

andy2012zwj, 2020-04-07:
确定os代码是3.0.0版本，且未修改，然后执行以下命令：
aos make clean
aos make distclean
aos make helloworld@developerkit -c config
aos make export-keil
最后在生成的keil工程里再编译试试

aktoey, 2020-04-08:
E:\IDM_Download\RTOS\AliOS-Things-3.0.0>aos make clean
aos-cube version: 0.3.11
信息: 用提供的模式无法找到文件。
信息: 用提供的模式无法找到文件。
Cleaning...
Done

E:\IDM_Download\RTOS\AliOS-Things-3.0.0>aos make clean
aos-cube version: 0.3.11
信息: 用提供的模式无法找到文件。
信息: 用提供的模式无法找到文件。
Cleaning...
Done

E:\IDM_Download\RTOS\AliOS-Things-3.0.0>aos make distclean
aos-cube version: 0.3.11
信息: 用提供的模式无法找到文件。
信息: 用提供的模式无法找到文件。
Cleaning...
Done

E:\IDM_Download\RTOS\AliOS-Things-3.0.0>aos make helloworld @ developerkit -c config
aos-cube version: 0.3.11
信息: 用提供的模式无法找到文件。
信息: 用提供的模式无法找到文件。
build/Makefile:327: warning: overriding recipe for target 'out/helloworld'
build/build_rules/aos_prebuild.mk:16: warning: ignoring old recipe for target 'out/helloworld'
build/Makefile:327: warning: overriding recipe for target '@'
build/build_rules/aos_prebuild.mk:16: warning: ignoring old recipe for target '@'
build/Makefile:330: warning: overriding recipe for target 'out/helloworld'
build/Makefile:327: warning: ignoring old recipe for target 'out/helloworld'
build/Makefile:330: warning: overriding recipe for target '@'
build/Makefile:327: warning: ignoring old recipe for target '@'
build/Makefile:336: warning: overriding recipe for target 'out/helloworld'
build/Makefile:330: warning: ignoring old recipe for target 'out/helloworld'
build/Makefile:336: warning: overriding recipe for target '@'
build/Makefile:330: warning: ignoring old recipe for target '@'
build/Makefile:397: warning: overriding recipe for target 'out/helloworld'
build/Makefile:336: warning: ignoring old recipe for target 'out/helloworld'
build/Makefile:397: warning: overriding recipe for target '@'
build/Makefile:336: warning: ignoring old recipe for target '@'
make.exe: Circular out/helloworld <- out/helloworld dependency dropped.
make.exe: Circular out/helloworld <- @ dependency dropped.
make.exe: Circular developerkit/auto_component/aos.mk <- out/helloworld dependency dropped.
make.exe: Circular developerkit/auto_component/aos.mk <- @ dependency dropped.
kconfig tools missing, start download ...
https://gitee.com/alios-things/kconfig-frontends-win32.git -> .//build/kconfig/Win32/
Cloning into './/build/kconfig/Win32'...
remote: Enumerating objects: 19, done.
remote: Counting objects: 100% (19/19), done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 19 (delta 7), reused 4 (delta 2), pack-reused 0
Unpacking objects: 100% (19/19), done.
Creating .defconfig ...
Creating .config ...
#
# configuration written to .config
#
Checking config output: .config ...
[ERROR]: The build string is invalid helloworld
make.exe: *** No rule to make target '.config', needed by 'out/config/autoconf.h'.  Stop.
[AliOS-Things] ERROR: "E:/IDM_Download/RTOS/AliOS-Things-3.0.0/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/IDM_Download/RTOS/AliOS-Things-3.0.0/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/IDM_Download/RTOS/AliOS-Things-3.0.0/build -e -f build/Makefile helloworld @.config developerkit" in "E:\IDM_Download\RTOS\AliOS-Things-3.0.0"
---

E:\IDM_Download\RTOS\AliOS-Things-3.0.0>aos make helloworld @ developerkit -c config
aos-cube version: 0.3.11
build/Makefile:327: warning: overriding recipe for target 'out/helloworld'
build/build_rules/aos_prebuild.mk:16: warning: ignoring old recipe for target 'out/helloworld'
build/Makefile:327: warning: overriding recipe for target '@'
[... 20 lines cut ...]
# configuration written to .config
#
Checking config output: .config ...
[ERROR]: The build string is invalid helloworld
make.exe: *** No rule to make target '.config', needed by 'out/config/autoconf.h'.  Stop.
[AliOS-Things] ERROR: "E:/IDM_Download/RTOS/AliOS-Things-3.0.0/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/IDM_Download/RTOS/AliOS-Things-3.0.0/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/IDM_Download/RTOS/AliOS-Things-3.0.0/build -e -f build/Makefile helloworld @.config developerkit" in "E:\IDM_Download\RTOS\AliOS-Things-3.0.0"
---

E:\IDM_Download\RTOS\AliOS-Things-3.0.0>aos make export-keil
aos-cube version: 0.3.11
Parsing all components ...
Creating out/config/autoconf.h ...
Making config file for first time
app_version:app-1.0.0-20200408.193316
processing components: helloworld developerkit
*** All Components: helloworld developerkit osal_aos mcu_stm32l4xx_cube kernel_init sensor sal netmgr rhino debug arch_armv7m newlib_stub ulog vfs device_sal_bk7231 kv yloop atparser lib_rbtree
*** Config Enabled: lib_rbtree
Making keil Project
Creating keil project helloworld@developerkit
Keil project created at projects/Keil/helloworld@developerkit/keil_project/helloworld@developerkit.uvprojx
----------- keil project has generated in projects/Keil/helloworld@developerkit/keil_project -----------
Build Keil project complete

E:\IDM_Download\RTOS\AliOS-Things-3.0.0>

closed 2022-03-01 by skylarCai