alibaba/AliOS-Things #1183: esp32 awss ble provision can not compile.

fumiot, 2020-05-28:
**Describe the bug**
When we enable FEATURE_AWSS_SUPPORT_BLE_PROVISON as we did under rel_2.1.0, build failed.

**environment**
1. branch(version): master(v3.1.0)
2. Desktop OS(windows linux or macOS):linux
3. example: 
linkkitapp
**Expected behavior**
build success , awss by ble success. We can do it in rel_2.1.0.

**Screenshots**
In file included from /home/p/fumiot/AliOS-Things/components/wireless/bluetooth/ble/host/bt_common/include/toolchain.h:21:0,
                 from /home/p/fumiot/AliOS-Things/components/wireless/bluetooth/ble/host/include/bluetooth/hci.h:11,
                 from /home/p/fumiot/AliOS-Things/include/wireless/bluetooth/bluetooth/bluetooth.h:25,
                 from /home/p/fumiot/AliOS-Things/components/wireless/bluetooth/ble/breeze/ref-impl/breeze_hal_ble.c:7:
/home/p/fumiot/AliOS-Things/components/wireless/bluetooth/ble/host/bt_common/include/toolchain/gcc.h:339:2: error: #error processor architecture not supported
 #error processor architecture not supported

**Additional context**

whwxxstsyd, 2021-07-21:
关于v3.1.0版本编译通不过的问题，先说怎么解决，只要在linkkitapp目录下的Config.in配置文件中加入这一段就可以解决

[image]

再说说是怎么发现并解决的，全局搜索出现异常的报错位置

[image]

找到对应的地方

[image]

一步步跟踪	

[image]

继续跟踪

[image]

再看一下breezeapp历程下的Config.in文件，里面对 EN_BLE_HOST 做了相关配置

[image]

报错的原因主要是没有指定BLE_HOST

whwxxstsyd, 2021-07-21:
还有一个v3.1.0版本可能遇到的坑也在这里做个记录
云智能APP收不到蓝牙配网的信号

Alios的蓝牙配网使用开源breeze项目

正常配网，云智能APP搜索到蓝牙信号，会弹出如下界面：

[image]

但是在v3.1.0版本上无法出现，但是手机上确实可以发现有蓝牙信号发出

正好手头有一块可用云智能APP配网的TG7100C开发版，所以就对两者进行抓包分析

TG7100C抓包如下：

[image]

ESP32抓包如下：

[image]

关于数据包的各个字段的意思，可以看下我做的笔记（字有点丑）

[image]

再从源码中分析一下

ESP32源码：

[image]

看一下ESP32 上的 breeze 的版本：
 
[image]

TG7100C源码：

[image]

看一下TG7100C 上的 breeze版本：

[image]

因为TG7100C和ESP32的广播数据差别只有在VID和FMSK不同，所以我的解决方法就是直接在源代码中直接写死参数。最后测试发现，能否被云智能识别到只和VID有关，猜测可能是Alios-rel_3.1.0分支的breeze版本太低，数据包被云智能过滤了，强制写死VID就能被发现了

closed 2022-03-01 by skylarCai