alibaba/AliOS-Things #1201: 2ndboot中flash_get_partiton函数判断下边界问题，会导致找不到分区

WilliamArthur, 2020-06-26:
**Describe the bug**
hal_main.c中flash_get_partiton中判断操作地址下边界
语句为：
        if ((addr >= partition->partition_start_addr) &&
            (addr < (partition->partition_start_addr + partition->partition_length)) &&
            ((addr + len) >= partition->partition_start_addr) &&
            ((addr + len) < (partition->partition_start_addr + partition->partition_length)))
其中下边界判断语句为
           ((addr + len) >= partition->partition_start_addr) &&
            ((addr + len) < (partition->partition_start_addr + partition->partition_length)))

(addr + len - 1)才应该是最后一个操作地址，如果(addr + len) >= partition->partition_start_addr)
等于成立，那最后操作的地址不应该是上一个分区了吗，另外
(addr + len) < (partition->partition_start_addr + partition->partition_length）左右两边各减去1等式仍然成立，
应该是最后一个操作地址<=分区最后一个地址才对。不然如果操作整个分区，根本就选不出分区来，正确应该为：
           ((addr + len - 1) >= partition->partition_start_addr) &&
            ((addr + len - 1) < (partition->partition_start_addr + partition->partition_length)))

**environment**
1. branch(version): alios things v3.1
2. Desktop OS(windows linux or macOS):windows
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

WilliamArthur, 2020-06-26:
[image]
ymodem_upgrade（）函数中有同样的问题：
   if((addr >= app_info->partition_start_addr) && (addr <= app_info->partition_start_addr + app_info->partition_length)) {
        flag = 1;
        ymodem_max_write_size = app_info->partition_length;
        ret = hal_flash_erase(HAL_PARTITION_APPLICATION, 0, app_info->partition_length);
        if(ret != 0) {
            printf("erase err.\n");
            return -1;
        }
    }
表达式addr <= app_info->partition_start_addr + app_info->partition_length，应该改为
addr < app_info->partition_start_addr + app_info->partition_length，
其中app_info->partition_start_addr + app_info->partition_length为下一个分区的首地址，本
分区的结束地址应该为app_info->partition_start_addr + app_info->partition_length - 1

andy2012zwj, 2020-06-29:
flash_get_partiton中如果写到分区的最后一块的话，会有问题，之前没有出现是因为没有写到分区的最后一块；

closed 2022-03-01 by skylarCai