alibaba/AliOS-Things #1267: 请问有支持js开发ui的计划吗？

camel007, 2021-01-12:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Additional context**

ElliottSeer, 2021-02-25:
我们已经有HaaS UI 的JS解决方案了，并且可供免费使用。更多信息可以加入钉钉群获取
[image]

FreshLuoBoGan, 2022-01-24:
> 我们已经有HaaS UI 的JS解决方案了，并且可供免费使用。更多信息可以加入钉钉群获取 [image]

这个群进不去，方便再发一个吗？

skylarCai, 2022-01-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
        您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)
        如果“HaaS百事通”没有解决您的问题，我们会在两个工作日回复您。

YiluMao, 2022-01-25:
哦，这个问题也是您提的呀，那我们就在另一个问题里面讨论吧

ElliottSeer, 2022-01-25:
> > 我们已经有HaaS UI 的JS解决方案了，并且可供免费使用。更多信息可以加入钉钉群获取 [image]
> 
> 这个群进不去，方便再发一个吗？

试试这个
[image]

closed 2022-03-01 by skylarCai