alibaba/AliOS-Things #1326: AliOS Things 3.3 轻松上手实践典籍

yaoqiuguo, 2021-06-15:
AliOS Things 3.3版本于2021年4月23日正式发布开源，项目地址：
   - Gitee: [https://gitee.com/alios-things](https://gitee.com/alios-things)
   - Github: [https://github.com/alibaba/AliOS-Things](https://github.com/alibaba/AliOS-Things)
   - Code China: [https://codechina.csdn.net/alios-things/AliOS-Things](https://codechina.csdn.net/alios-things/AliOS-Things)

本次发布包含的新特性有：
   - **升级AliOS Things操作系统功能**
      新增内核对象管理功能、优化驱动框架和AOS内核接口、增强调试能力。
   - **增加轻应用开发框架**
      支持JavaScript和Python两种语言开发AliOS Things应用。
   - **增强AliOS Things操作系统配套的开发工具**
      开发环境支持界面化应用开发、升级构建工具和资源配置方式、自研在线分析工具。SmartTraceV1.4发布以及增加对通用分析工具SystemView的支持。
   - **更新AliOS Things操作系统及配套开发工具的说明文档**
      提供更详尽，更易读的中文文档。
      文档与操作系统源代码同时开源，开放与开发者的共建文档资源库的窗口。
   - **扩充AliOS Things操作系统开发者生态**
      推出CSDN和微信技术社区、钉钉群以及HaaS百事通等通道，全方位的服务开发者。

我们会继续秉持全面开源的思想，不断把新开发出来的组件和案例，快速的贡献到开源社区，请大家持续关注。

# AliOS Things 3.3 开源仓库

AliOS Things 3.3版本分别在[Gitee](https://gitee.com/alios-things)、[Github](https://github.com/alibaba/AliOS-Things/tree/rel_3.3.0)、[Code China](https://codechina.csdn.net/alios-things/AliOS-Things/-/tree/rel_3.3.0)开源，分支均为rel_3.3.0。用户贡献代码可在Gitee上提交申请，AliOS Things开发工具默认是从[Gitee](https://gitee.com/alios-things)下载源码。[Github](https://github.com/alibaba/AliOS-Things/tree/rel_3.3.0)、[Code China](https://codechina.csdn.net/alios-things/AliOS-Things/-/tree/rel_3.3.0)主要是方便用户浏览所有开源代码。

## Gitee

AliOS Things 3.3版本在Gitee平台上，每个功能模块分配一个独立的仓库。概览页面的精选区域目前是放置着常用的案例仓库和通用文档仓库，如简单示例工程helloworld_demo、通用文档仓库documentation等，如下图。

<div align="left">
  [image]
</div>

若用户想查看或搜索其他组件或案例仓库，可以点击上图选项栏中的“仓库”切换到“仓库”页，如下图。在搜索栏中输入目标组件或案例的名称，确认后即可显示搜索结果。

<div align="left">
  [image]
</div>

打开目标组件或案例的仓库，切换代码分支为rel_3.3.0，即可查看到该组件或案例3.3版本对应的源代码，如下图。

<div align="left">
  [image]
</div>

## Github
AliOS Things 3.3版本在Github平台上，所有功能模块共用一个仓库。用户可以通过登录本项目在Github上的地址，并切换代码分支为rel_3.3.0，即可查看到AliOS Things 3.3 版本所有开源代码，如下图。

<div align="left">
  [image]
</div>

## Code China
AliOS Things 3.3版本在Code China平台上，所有功能模块共用一个仓库。用户可以通过登录本项目在Code China上的地址，并切换代码分支为rel_3.3.0，即可查看到AliOS Things 3.3 版本所有开源代码，如下图。

<div align="left">
  [image]
</div>

# AliOS Things 3.3 系统概述
## 系统架构
AliOS Things 3.3版本系统架构如下图：

   - 硬件支持HaaS生态的HaaS100系列、HaaS EDU系列、HaaS200系列开发板；CPU架构支持ARM；内核提供调度、任务管理、内存管理等基础功能；组件层提供支持Native应用开发和HaaS轻应用开发的多种功能组件；系统增强了设备端应用与云服务的交互能力。
   - 为了降低物联网开发的门槛，我们提供了轻应用的开发框架：提供支撑[轻应用](https://help.aliyun.com/document_detail/174810.html)运行所需的全套基础设施，是一套集成了包括硬件抽象、网络接口、物联网组件、运行时引擎、对象封装等功能的应用软件框架。
   - 为提高开发者基于AliOS Things进行应用开发的效率，本次发布对AliOS Things配套的开发工具进行了全面的升级和增强。推出针对轻应用的开发工具、升级了AliOS Studio和构建系统、新增SytemView和SmartTrace工具辅助开发者调试分析。

系统架构图中，使用**深灰色**和**橙色**字体标记的模块是经过测试验证并已开源的模块，如“OTA组件”、“JS引擎”；而使用**浅灰色**字体标记的模块是正在研发或待测试的模块，如“支付组件”，今后会陆续开源。

<div align="left">
  [image]
[... 204 lines cut ...]

  对内核和组件常用API的提供说明。

## 访问入口

- [AliOS Things新版文档库](https://haas.iot.aliyun.com/aliosthings/)
- [AliOS Things文档中心](https://help.aliyun.com/product/123206.html?spm=a2c4g.750001.list.243.b1767b13EzBsGE)，选择“**AliOS Things 文档库（新）**”进入[AliOS Things新版文档库](https://haas.iot.aliyun.com/aliosthings/)
- [HaaS官网](https://haas.iot.aliyun.com/)，点击“**学习路径**”菜单即可进入[AliOS Things新版文档库](https://haas.iot.aliyun.com/aliosthings/)
- [阿里云官网](https://www.aliyun.com/)，点击主页右上方的“**文档**”菜单进入[阿里云文档中心](https://help.aliyun.com/?spm=5176.19720258.J_8058803260.30.e9392c4a3Zv7fb)，找到“**物联网**”板块，选择“**IoT物联网操作系统**”即可进入[AliOS Things文档中心](https://help.aliyun.com/product/123206.html?spm=a2c4g.750001.list.243.b1767b13EzBsGE)

# AliOS Things 开发者生态
## 开发者贡献

- [贡献代码](https://g.alicdn.com/alios-things-3.3/doc/contribute_code.html)
- [贡献文档](https://g.alicdn.com/alios-things-3.3/doc/contribute_doc.html)
- [贡献HaaS积木](https://haas.iot.aliyun.com/solution)
## 开发者支持

- HaaS官网：[https://haas.iot.aliyun.com/](https://haas.iot.aliyun.com/)
- HaaS技术社区：[https://blog.csdn.net/HaaSTech](https://blog.csdn.net/HaaSTech)​
- 开发者钉钉群和微信公众号见下图，开发者钉钉群每天都有技术支持同学值班。

<div align="left">
  [image]
</div>

closed 2022-07-20 by YiluMao

Pillar1989, 2024-01-19:
https://m.gelonghui.com/p/679272  AliOS-Things以后是什么路线呢 ？ 还搞吗 ？

Pillar1989, 2024-01-19:
@YiluMao