alibaba/AliOS-Things #1342: 新建工程的haas1000_write_flash_main.exe闪退

qxdn, 2021-11-16:
新建工程后打开haas1000_write_flash_main.exe时闪退。

`git clone https://github.com/alibaba/AliOS-Things.git -b dev_3.1.0_haas`这个版本的flash下载工具正常，希望能够修复一下

jinliangli, 2021-11-18:
亲， 能提供一下你的运行环境吗？比如windows版本，操作步骤，和出现闪退的概率。

我在windows 10 64bit/windows 7 64 bit 上运行 haas1000_write_flash_main.exe均不能复现。

jinliangli, 2021-11-18:
另外，为安全起见， 推荐使用HaaS Studio的烧录 https://help.aliyun.com/document_detail/302383.html。 只有需要更新haas1000的bootloader的时候才需要使用haas1000_write_flash_main.exe。

qxdn, 2021-11-18:
- OS： win10 21H1  19043.1348
- 操作步骤： 新建工程，编译后双击`haas1000_write_flash_main.exe`
- 闪退频率： 每次

[image]

问题的详细信息见上，使用haas1000_write_flash_main.exe主要还是因为出厂固件有问题，无法使用haas studio烧录

ps:
- 执行clean命令时有`/usr/bin/bash: -c: line 1: syntax error: unexpected end of file`问题。makeflie开头加上`SHELL := C:\Windows\System32\cmd.exe`解决
- 新建工程的目录不能有空格，编译会出错

qxdn, 2021-11-18:
gitee仓库那边缺少了`productline_cfg.xml`文件，将该文件放入write_flash_gui文件夹后haas1000_write_flash_main.exe就正常了

jinliangli, 2021-11-19:
我们鼓励开发者为AliOS Things贡献代码， 所以你方便为这个问题提交个pull request吗， 只需要按照这个https://haas.iot.aliyun.com/aliosthings/contribute_code.html 操作即可， 提交到https://gitee.com/alios-things/haas1000 仓库。

Patch会保留你的作者信息，先谢谢你的积极参与。

jinliangli, 2021-11-19:
> ps:
> 
> * 执行clean命令时有`/usr/bin/bash: -c: line 1: syntax error: unexpected end of file`问题。makeflie开头加上`SHELL := C:\Windows\System32\cmd.exe`解决
> * 新建工程的目录不能有空格，编译会出错
你反馈的这两个问题，我会找负责IDE的同事确认，感谢你的反馈。 另外后续不同的问题，可以单独开一个issue，方便问题的跟踪管理。谢谢你的积极参与。