alibaba/AliOS-Things #1347: 支持ccs控制UI吗？我可以按照那个demo来测试？

FreshLuoBoGan, 2022-01-24:


skylarCai, 2022-01-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
        您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)
        如果“HaaS百事通”没有解决您的问题，我们会在两个工作日回复您。

YiluMao, 2022-01-24:
请问CCS是什么？

FreshLuoBoGan, 2022-01-24:
抱歉，css。我想了解一下HaasUI那一部分。

FreshLuoBoGan, 2022-01-24:
[image]

YiluMao, 2022-01-25:
我们目前是支持的，但这部分还没开源
请问您这边的需求是什么？

FreshLuoBoGan, 2022-01-25:
我想了解一下这部的前端开发语言是什么？是vue吗？我们嵌入式设备是海思的，我看是POSIX，是否可以无缝迁移；

YiluMao, 2022-01-25:
是JS

FreshLuoBoGan, 2022-01-25:
哦，那我再请教件事哈，是js控制native的c++组件吗？我们公司有一堆的前端组件，这种情况是不是没办法复用。

YiluMao, 2022-01-25:
你是说你们公司有一堆native的组件是吗？

FreshLuoBoGan, 2022-01-25:
我们公司有一堆的前端组件。native组件也有是，是skia的。

YiluMao, 2022-01-25:
https://haas.iot.aliyun.com/haasui?spm=a2cpu.b50127119.0.0.595660b1kutjPI

FreshLuoBoGan, 2022-01-26:
这个内容我看过，体验了一下模拟器，但是我有点无从下手，我想在海思上体验一下，我看好像是POSIX，应该和不区分主芯片吧（是走的frame buffer吧）？这个我咋体验一下？

YiluMao, 2022-01-27:
我给你找人支持

liuwenxin666, 2022-01-27:
是走的frame buffer，如果要在海思上体验，您需要提供包含framebuffer送显接口（及触摸接口）的动/静态库，HaaS UI会把绘制好的图像buffer输送出来，传给您提供的送显接口去显示，需要对接一下送显和触摸的接口即可，具体如下：
extern "C" void uinput_callback(uinput_event_type_t event, void* pdata, uint32_t len)
extern "C" void display_draw(void* buffer);

> 这个内容我看过，体验了一下模拟器，但是我有点无从下手，我想在海思上体验一下，我看好像是POSIX，应该和不区分主芯片吧（是走的frame buffer吧）？这个我咋体验一下？

FreshLuoBoGan, 2022-01-28:
uinput_event_type_t

FreshLuoBoGan, 2022-01-28:
> 是走的frame buffer，如果要在海思上体验，您需要提供包含framebuffer送显接口（及触摸接口）的动/静态库，HaaS UI会把绘制好的图像buffer输送出来，传给您提供的送显接口去显示，需要对接一下送显和触摸的接口即可，具体如下： extern "C" void uinput_callback(uinput_event_type_t event, void* pdata, [...]

1. uinput_event_type_t  我没找到这个的定义，这个请教一下如何定义的。
2. display_draw 我们有现成的，是c接口。

我这个封装成一个库后，再如何处理。

liuwenxin666, 2022-01-28:
> 我想了解一下这部的前端开发语言是什么？是vue吗？我们嵌入式设备是海思的，我看是POSIX，是否可以无缝迁移；

是vue

FreshLuoBoGan, 2022-01-28:
@

> 1. uinput_event_type_t  我没找到这个的定义，这个请教一下如何定义的。
> 2. display_draw 我们有现成的，是c接口。
>
> 我这个封装成一个库后，再如何处理。

@liuwenxin666 麻烦再看一下这个

liuwenxin666, 2022-01-28:
如果要对接新的设备，请您等haas ui后续推出推广版本后再来对接，届时我们会给出具体的对接方案，感谢关注

liuwenxin666, 2022-01-28:
> @
>
>
> @liuwenxin666 麻烦再看一下这个

如果要对接新的设备，请您等haas ui后续推出推广版本后再来对接，届时我们会给出具体的对接方案，感谢关注

FreshLuoBoGan, 2022-01-28:
再麻烦问一下，后续的推广版本有时间计划吗？ @liuwenxin666

liuwenxin666, 2022-01-28:
> 再麻烦问一下，后续的推广版本有时间计划吗？ @liuwenxin666

这个目前还在规划中，有消息及时通知您