alibaba/AliOS-Things #1348: 进程间通信，可以参考哪里？

FreshLuoBoGan, 2022-01-28:
https://haas.iot.aliyun.com/aliosthings/index.html

我在这个指导文档中没有找到进程间通信的接口，但是我看图里面有，麻烦请教一下，我可以参考哪里。

skylarCai, 2022-01-28:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
        您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)
        如果“HaaS百事通”没有解决您的问题，我们会在两个工作日回复您。

YiluMao, 2022-01-28:
你可以看看我们微内核分支的代码
目前我们3.3还没有支持进程，所以这部分代码是没有的

YiluMao, 2022-01-28:
另外，有没有兴趣加入我们？
https://talent.alibaba.com/off-campus-position/727473?trace=qrcode_share

可以把简历给我哈

FreshLuoBoGan, 2022-01-28:
> 你可以看看我们微内核分支的代码 目前我们3.3还没有支持进程，所以这部分代码是没有的

微内核分支代码麻烦引导一下。

FreshLuoBoGan, 2022-01-28:
> 另外，有没有兴趣加入我们？ https://talent.alibaba.com/off-campus-position/727473?trace=qrcode_share
> 
> 可以把简历给我哈

skylarCai, 2022-02-08:
> > 你可以看看我们微内核分支的代码 目前我们3.3还没有支持进程，所以这部分代码是没有的
> 
> 微内核分支代码，我去哪里可以看到，

请切换到[rel_microkernel](https://github.com/alibaba/AliOS-Things/tree/rel_microkernel) 分支，此分支为微内核代码

closed 2022-03-02 by skylarCai