alibaba/AliOS-Things #1354: haas100开发板连接Wi-Fi总是断开连接然后快速重连

duanshuai007, 2022-02-22:
连接Wi-Fi的时候非常频繁的断开连接，然后又快速的重新连上。

qisheng-seu, 2022-02-22:
亲，您可以按照以下步骤排查下，
1、确认天线是否正确连接。
2、CLI中敲入 aos_mac WIFI 命令查询下HaaS100的wifi mac地址是否正确设置，您也可以通过路由器端查询连接上的HaaS100的mac地址是否合法。
3、确认HaaS100和路由器的距离是否在5米以内，太远的距离可以导致wifi连接不稳定。
[image]

duanshuai007, 2022-02-22:
1，2，3都分别确认了，没有问题。
但还是会出现掉线的情况。

skylarCai, 2022-02-22:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

duanshuai007, 2022-02-22:
人工支持

qisheng-seu, 2022-02-22:
亲，针对您的问题，我们已经紧急请求芯片合作厂商帮忙一起排查。现在需要您辛苦帮忙提供问题全程的串口日志，谢谢！！

duanshuai007, 2022-02-22:
4256775/cw1200_bh | [WSM] Issue unjoin command (RX).

  4256776/net_wq | [STA] Unjoin.

  4256776/net_wq | wlan0: deauthenticated from 0x2005a21cM (Reason: 3)

  4256776/net_wq | ieee80211_set_disassoc 1 0

  4256779/net_wq | cw1200_bss_info_changed=0x2000 hw_priv:0x200394c0 priv:0x2003a208

  4256779/net_wq | [STA] Freq 2437 (wsm ch: 6).

  4256779/net_wq | cw1200_bss_info_changed=0x109f hw_priv:0x200394c0 priv:0x2003a208

  4256779/net_wq | [STA] BSS_CHANGED_ARP_FILTER enabled: 0, cnt: 1

  4256779/net_wq | BSS_CHANGED_ASSOC.

  4256779/net_wq | [STA] Slot time :20 us.

  4256779/net_wq | [CQM] RSSI threshold subscribe: 0 +- 0

  4256779/net_wq | [CQM] Beacon loss subscribe: 0

  4256779/net_wq | [CQM] TX failure subscribe: 0

  4256780/net_wq | cw1200_bss_info_changed=0x4000 hw_priv:0x200394c0 priv:0x2003a208

  4256780/net_wq | ignore IEEE80211_CONF_CHANGE_MONITOR (0) IEEE80211_CONF_CHANGE_IDLE (1)

  4256780/net_wq | nl80211_send_deauth

  4256780/net_wq | bwifi send event, id:3
  4256780/wifi_even | [WIFI] status CONNECTED, event 3
  4256780/wifi_even | disconnected from AP
  4256780/wifi_even | on_wifi_disconnect event_id:3
  4256780/wifi_even | wifi disconnected (reason=3, locally_generated=0)
  4256780/wifi_even | [WIFI] change status:CONNECTED->IDLE
[4259.567]<E>WIFI_SERVICE wifi_connect_failed_cb:300 err=4124

[4259.572]<E>NETMGR_CONN netmgr_conn_state_change:188 old state=5 new state=6

[4259.573]<E>NETMGR_CONN conn_failed:144 wrong state=5

---> wifi disconnect
---> wifi disconnect
01 06 20 0f a0 00 a0 00 00 00 00 00 00 00 00 00 00 07 00 
04 0e 04 05 06 20 00 
01 08 20 20 14 03 03 0f ff 07 09 7a 79 63 75 62 65 07 ff 2c 16 bd d0 12 a6 00 00 00 00 00 00 00 00 00 00 00 
04 0e 04 05 08 20 00 
01 09 20 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
04 0e 04 05 09 20 00 
01 0a 20 01 01 
04 0e 04 05 0a 20 00 
---- wifi state = 1
[4264.575]<E>WIFI_SERVICE get_wifi_config:1111 open /data/temp/temp.conf failed:No such file or directory

[4264.577]<E>WIFI_SERVICE ssid_contain_chinese_check:2191 Error: app ssid is NULL!

[4264.577]<E>WIFI_SERVICE netmgr_wifi_connect:2395 no config found

  4261789/m_wifi_ta | Scan for specific SSID Frog
  4261789/m_wifi_ta | [WIFI] change status:IDLE->SCANNING
  4261789/m_wifi_ta | nl80211: request->n_ssids=1, request->n_channels=13
  4261789/m_wifi_ta | cw1200_bss_info_changed=0x4000 hw_priv:0x200394c0 priv:0x2003a208

  4261789/m_wifi_ta | ignore IEEE80211_CONF_CHANGE_MONITOR (0) IEEE80211_CONF_CHANG  4261947/net_taskl | ieee80211_scan_rx: find qualified SSID in ProbeResp (2437,"Frog",-30,"80:8f:1d:e6:6a:4c")
  4261948/net_taskl | [SCAN] Cancel scan in progress

  4261948/cw1200_bh | cw1200_scan_complete_cb status:0
[... 150 lines cut ...]
  4262178/net_taskl | EAPOL: SUPP_BE entering state IDLE
  4262178/net_taskl | eap_peer_sm_step 1485.
  4262178/net_taskl | eap_peer_sm_step 1485.
  4262178/wifi_even | [WIFI] status CONNECTING, event 1
  4262178/wifi_even | wifi connecting in state 9
  4262179/wifi_even | on_wifi_connect event_id:0
  4262179/wifi_even | [WIFI] status CONNECTING, event 2
  4262179/wifi_even | wifi mac connected to Frog
  4262179/wifi_even | on_wifi_connect event_id:1
  4262179/wifi_even | [WIFI] change status:CONNECTING->CONNECTED
  4262179/wifi_even | wsm_epta_cmd 60000 40000 3.
  4262179/wifi_even | wsm_epta_cmd 60000 40000 0.
  4262179/wifi_even | Set PS mode 0
  4262179/wifi_even | nl80211_set_power_save disabled
  4262179/wifi_even | config_dpd, ch=6
  4262180/wifi_conn | connect complete, success
  4262244/tcpip_thr | set ip: 192.168.200.71
  4262244/tcpip_thr | set mask: 255.255.255.0
  4262244/tcpip_thr | set gw: 192.168.200.252
  4262244/tcpip_thr | cw1200_bss_info_changed=0x1000 hw_priv:0x200394c0 priv:0x2003a208

  4262244/tcpip_thr | [STA] BSS_CHANGED_ARP_FILTER enabled: 1, cnt: 1

  4262244/tcpip_thr | wsm_epta_cmd 60000 40000 0.
---> wifi dhcp success

duanshuai007, 2022-02-22:
这部分日志是在正常工作中，然后网络断开，再到重新连接到网络的部分日志信息

qisheng-seu, 2022-02-22:
亲，感谢您的日志反馈，我们工程师正在紧锣密鼓排查中，会尽快给您回复。谢谢

qisheng-seu, 2022-02-23:
从您提供的日志看，附近存在2个名为Frog的路由器，请问这两个路由器的加密方式和password都一样吗？看起来和2个同名路由器有关。可以尝试换个不同名字的路由器试试。
4261948/net_wq | scan result0:(2437,"Frog",-33,"80:8f:1d:ef:f1:21")
4261948/net_wq | scan result1:(2437,"Frog",-30,"80:8f:1d:e6:6a:4c")

duanshuai007, 2022-02-23:
是的，存在4个同名路由器。但是设备已经连接上路由器了，为什么会断开呢，源码会将这个问题视为一个bug并进行修复吗？

qisheng-seu, 2022-02-23:
看日志是路由器端把设备踢掉的。要继续排查的话需要您提供以下几点信息，
1、换个路由器名字是否还会出现该问题？
2、4个同名路由器的加密方式和密码是否都是一样的？

duanshuai007, 2022-02-23:
好的，我去试一试。

duanshuai007, 2022-02-23:
测试了一个多小时。连接我手机开启的热点没掉线。
那这个问题也还是存在的啊，在实际环境中还是有可能出现多个路由器是同样的wifi名和密码的。
最主要是我现在环境下用esp32就没有这个问题，只有haas100开发板有这个问题。

qisheng-seu, 2022-02-23:
感谢您的反馈。我们内部排查下，晚点给您回复。

qisheng-seu, 2022-02-23:
请帮忙提供下路由器的型号，我们看下是否有相同路由器复现下？另外你那边是否可以抓wifi air sniffer日志 ？

duanshuai007, 2022-02-23:
路由器是TP-LINK的TL-WDR7650千兆易展版（AC1900双频千兆无线路由器，在范围内有4个这样的路由器，都具有相同的SSID和PSK）,我这里看不到路由器端的日志。

qisheng-seu, 2022-02-23:
好的。我将信息反馈给合作方，有解决方案时会第一时间通知您。谢谢

closed 2022-03-07 by skylarCai