alibaba/AliOS-Things #1359: 关于haas100以连接WiFi后进行数据传输时，在串口遇到的问题，麻烦尽快解决，辛苦啦

HRO547, 2022-02-23:
[Ne w File.txt](https://github.com/alibaba/AliOS-Things/files/8122273/Ne.w.File.txt)

skylarCai, 2022-02-23:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-02-23:
看起来已经连上了，请问您的问题是什么？

HRO547, 2022-02-23:
> 看起来已经连上了，请问您的问题是什么？

连是连上了，但是为什么阿里云的物联网平台的设备还是离线状态呢？（已经烧录完成且成功过，但现在更换WiFi就不行了）

HRO547, 2022-02-23:
> 看起来已经连上了，请问您的问题是什么？

aiot_mqtt_connect failed: -0x0F23这个该怎么处理呢？

YiluMao, 2022-02-23:
稍等

YiluMao, 2022-02-23:
#define STATE_PORT_TLS_INVALID_HANDSHAKE                            (-0x0F23)

YiluMao, 2022-02-23:
也可以看看板子的天线有没有连好

HRO547, 2022-02-23:
> #define STATE_PORT_TLS_INVALID_HANDSHAKE （-0x0F23）
> 
> 可以检查下wifi的稳定性

如果是这个原因的话解决方案不就只能跟换WiFi了吗？

yanxiaoyong-ch, 2022-02-23:
> 如果是这个原因的话解决方案不就只能跟换WiFi了吗？

1.检查下热点，看是不是稳定；2.检查天线；3.发一下完整的log

HRO547, 2022-02-23:
[New File.txt](https://github.com/alibaba/AliOS-Things/files/8122464/New.File.txt)
进行配网之后就一直重复下面的那段

yanxiaoyong-ch, 2022-02-23:
> [New File.txt](https://github.com/alibaba/AliOS-Things/files/8122464/New.File.txt) 进行配网之后就一直重复下面的那段

请问用的哪个solution？

HRO547, 2022-02-23:
> 

智能养花检测温湿度那个

yanxiaoyong-ch, 2022-02-23:
flower_demo/data_model_basic_demo.c demo_main.c里面的三元组有替换成你自己的吗

yanxiaoyong-ch, 2022-02-23:
> > flower_demo/data_model_basic_demo.c demo_main.c里面的三元组有替换成你自己的吗
> 
> 有的，之前用的WiFi是可以的，但是在更换WiFi之后就一直不行了

"之前用的WiFi是可以的，但是在更换WiFi之后就一直不行了",的意思原来连接另一个wifi热点时是ok的吗？

HRO547, 2022-02-23:
> > > flower_demo/data_model_basic_demo.c demo_main.c里面的三元组有替换成你自己的吗
> > 
> > 
> > 有的，之前用的WiFi是可以的，但是在更换WiFi之后就一直不行了
> 
> "之前用的WiFi是可以的，但是在更换WiFi之后就一直不行了"，的意思原来连接另一个wifi热点时是ok的吗？

是的，但是试了好多个WiFi就只有一个是好的，haas100的WiFi连接的要求很高吗？

yanxiaoyong-ch, 2022-02-23:
> > > > flower_demo/data_model_basic_demo.c demo_main.c里面的三元组有替换成你自己的吗
> > > 
> > > 
> > > 有的，之前用的WiFi是可以的，但是在更换WiFi之后就一直不行了
> > 
> > 
> > "之前用的WiFi是可以的，但是在更换WiFi之后就一直不行了"，的意思原来连接另一个wifi热点时是ok的吗？
> 
> 是的，但是试了好多个WiFi就只有一个是好的，haas100的WiFi连接的要求很高吗？
天线有接上吗

HRO547, 2022-02-23:
> > > > > flower_demo/data_model_basic_demo.c demo_main.c里面的三元组有替换成你自己的吗
> > > > 
> > > > 
> > > > 有的，之前用的WiFi是可以的，但是在更换WiFi之后就一直不行了
> > > 
> > > 
> > > "之前用的WiFi是可以的，但是在更换WiFi之后就一直不行了"，的意思原来连接另一个wifi热点时是ok的吗？
> > 
> > 
> > 是的，但是试了好多个WiFi就只有一个是好的，haas100的WiFi连接的要求很高吗？
> > 天线有接上吗

有，确定是接好了的

yanxiaoyong-ch, 2022-02-23:
> 是

有问题的路由器型号能发下吗

closed 2022-03-16 by skylarCai