alibaba/AliOS-Things #1362: esp32 固件下载不了

rualdeng, 2022-02-24:
https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/startup/ESP32_startup  esp32 固件列表点击下载不了。

skylarCai, 2022-02-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yong171966, 2022-02-24:
https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16145223.0.0.595660b1B2CgrL#/Python/docs/zh-CN/startup/ESP32_startup
[image]
验证完成，是可以下载的亲！

skylarCai, 2022-02-24:
人工支持

zy20220224, 2022-02-24:
> https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/startup/ESP32_startup esp32 固件列表点击下载不了。

刚才验证可以正常下载，你在尝试一下。

closed 2022-03-02 by skylarCai