alibaba/AliOS-Things #1369: 乐鑫ESP32 import oss报错

rualdeng, 2022-02-24:
[OSS - OSS对象存储](https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/haas_extended_api/oss?id=oss-oss%e5%af%b9%e8%b1%a1%e5%ad%98%e5%82%a8)
import oss
报错：
Traceback (most recent call last):
  File "/data/pyamp/main.py", line 45, in <module>
ImportError: no module named 'oss'

import OSS(大写)也错同样的错误

# test.jpg存放在“/”目录下

file = oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, "test.jpg")

同时能否像之前的OSS一样设置object name，类似objectname = '/record/ttx.wav'，自定义上传后的路径

skylarCai, 2022-02-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

rualdeng, 2022-02-24:
固件：[HaaSPython-ESP32-v1.0.2](https://hli.aliyuncs.com/p/config/HaaS_Python/HaaSPython-ESP32-v1.0.2.zip)

QingjiangXu, 2022-02-25:
固件：[HaaSPython-ESP32-v1.0.2](https://hli.aliyuncs.com/p/config/HaaS_Python/HaaSPython-ESP32-v1.0.2.zip)
是针对NodeMCU-32的固件，由于存储空间限制没有打开oss.  
您可以在平台上使用help('modules')命令查看支持的module.

如果想使用oss，可以选择自己打开oss，并编译ESP32固件，方法如下：

rualdeng, 2022-02-25:
> 固件：[HaaSPython-ESP32-v1.0.2](https://hli.aliyuncs.com/p/config/HaaS_Python/HaaSPython-ESP32-v1.0.2.zip) 是针对NodeMCU-32的固件，由于存储空间限制没有打开oss，之后的M5Stack固件会默认打开； 您可以在平台上使用help('modules')命令查看支持的module.
> 
> 如果想使用oss，可以选择自己通过源代码打开oss，并编译ESP32固件，方法如下：
> 
> 1. 选择创建c 解决方案：py_engine_esp32_demo;
> 2. 在其中的package.yaml的def_congfig中，增加一行MICROPY_PY_OSS: 1

没办法创建C的解决方案，选择C语言支持的硬件只有haasedu,haas100,haas200。也没有py_engine_esp32_demo这个解决方案

QingjiangXu, 2022-03-02:
需要下载alios things的全部源代码，py_engine_esp32_demo在solution/下

closed 2022-03-07 by skylarCai