alibaba/AliOS-Things #1370: ESP32 audio - 音频系统模块 请把录音这个模块补上

rualdeng, 2022-02-24:
固件：[HaaSPython-ESP32-v1.0.2](https://hli.aliyuncs.com/p/config/HaaS_Python/HaaSPython-ESP32-v1.0.2.zip)
[audio - 音频系统模块](https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/haas_extended_api/audio?id=audio-%e9%9f%b3%e9%a2%91%e7%b3%bb%e7%bb%9f%e6%a8%a1%e5%9d%97) 
只有播放模块，没有录音模块。现需要录音功能，请把录音模块补上，谢谢！

skylarCai, 2022-02-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yong171966, 2022-02-25:
录音的功能已经存在了，文档上没有写。参考下面的案例使用：
from audio import  Recorder as recorder
>>> aRecorder=recorder()
>>> aRecorder.start('/data/1.amr', recorder.AMR)
True
I (516270) AMRNB_ENCODER: amrnb open
>>>
>>> aRecorder.stop()
I (527690) AMRNB_ENCODER: amrnb close
True

closed 2022-03-07 by skylarCai