alibaba/AliOS-Things #1375: import oss报错，没办法创建C解决方案，没有py_engine_esp32_demo这个解决方案

rualdeng, 2022-02-25:
固件：[HaaSPython-ESP32-v1.0.2](https://hli.aliyuncs.com/p/config/HaaS_Python/HaaSPython-ESP32-v1.0.2.zip)
是针对NodeMCU-32的固件，由于存储空间限制没有打开oss，之后的M5Stack固件会默认打开；
您可以在平台上使用help('modules')命令查看支持的module.

如果想使用oss，可以选择自己通过源代码打开oss，并编译ESP32固件，方法如下：
1. 选择创建c 解决方案：py_engine_esp32_demo;
2. 在其中的package.yaml的def_congfig中，增加一行MICROPY_PY_OSS: 1

_Originally posted by @QingjiangXu in https://github.com/alibaba/AliOS-Things/issues/1369#issuecomment-1050469387_

没办法创建C的解决方案，选择C语言支持的硬件只有haasedu,haas100,haas200。也没有py_engine_esp32_demo这个解决方案

skylarCai, 2022-02-25:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

rualdeng, 2022-02-25:
请提供一下如何自己编译ESP32固件的方法，以后也是需要自主去编译的

QingjiangXu, 2022-02-25:
编译ESP32固件，需要拉取AliOS-Things的原代码，然后对应解决方法的路径
https://github.com/alibaba/AliOS-Things/tree/master/solutions/py_engine_esp32_demo

QingjiangXu, 2022-02-25:
另外可以在开发者群里面获得详细支持

QingjiangXu, 2022-02-25:
也可以参考如下视频教程：https://www.bilibili.com/video/BV1gT4y197fA/

rualdeng, 2022-02-25:
> 也可以参考如下视频教程：https://www.bilibili.com/video/BV1gT4y197fA/

好的，我先看看，有什么问题再向你们咨询

closed 2022-03-15 by skylarCai