alibaba/AliOS-Things #1376: 编译ESP32固件，报错

rualdeng, 2022-02-25:
编译ESP32固件，需要拉取AliOS-Things的原代码，然后对应解决方法的路径
https://github.com/alibaba/AliOS-Things/tree/master/solutions/py_engine_esp32_demo

_Originally posted by @QingjiangXu in https://github.com/alibaba/AliOS-Things/issues/1375#issuecomment-1050614414_

编译 modemcu32的  py_engine_esp32  demo报这个错是啥情况啊？

change dir to E:\Program\IOT\AliOS\Project\Pratice\AliOS-Things\solutions\py_engine_esp32_demo.
Build Solution by nodemcu32s
scons: Reading SConscript files ...
There is no depend component `little_ui` found!
========
[{'little_ui': 'master ? <MICROPY_PY_LVGL>'}]
1
Do not reverse value
lost component check : no need install this component
[]
[WARNING] E:\Program\IOT\AliOS\Project\Pratice\AliOS-Things\components\py_engine\adapter/esp32/build-GENERIC/ is not exists or not 
directory.
[WARNING] E:\Program\IOT\AliOS\Project\Pratice\AliOS-Things\solutions\py_engine_esp32_demo\esp_sdk/include is not exists or not dir
ectory.
[WARNING] E:\Program\IOT\AliOS\Project\Pratice\AliOS-Things\solutions\py_engine_esp32_demo\esp_sdk/lib is not exists or not directo
ry.
[WARNING] E:\Program\IOT\AliOS\Project\Pratice\AliOS-Things\components\py_engine\adapter/esp32/build-GENERIC/ is not exists or not 
directory.
[WARNING] E:\Program\IOT\AliOS\Project\Pratice\AliOS-Things\solutions\py_engine_esp32_demo\esp_sdk/include is not exists or not dir
ectory.
[WARNING] E:\Program\IOT\AliOS\Project\Pratice\AliOS-Things\solutions\py_engine_esp32_demo\esp_sdk/lib is not exists or not directo
ry.
install win_longcmd_spawn patch in windows. 
文件名、目录名或卷标语法不正确。
Run script '"E:\Program\IOT\AliOS\Project\Pratice\AliOS-Things\components\py_engine\build_cmake.sh" "E:\Program\IOT\AliOS\Project\Pratice\AliOS-Things\solutions\py_engine_esp32_demo\out\py_engine_prebuild.txt" ' success
scons: done reading SConscript files.
scons: Building targets ...        
AR aos_sdk\lib\libespressif_esp32.a
RANLIB aos_sdk\lib\libespressif_esp32.a
CC out\py_engine\adapter\esp32\main.o
CC out\py_engine\external\amp_task\amp_task.o
命令行太长。
命令行太长。
CC out\py_engine\external\amp_boot\amp_boot.o
CC out\py_engine\external\amp_boot\amp_boot_cmd.o
命令行太长。
命令行太长。
CC out\py_engine\external\amp_boot\amp_boot_file_transfer.o
CC out\py_engine\external\amp_boot\amp_boot_uart.o
CC out\py_engine\external\amp_boot\amp_boot_ymodem.o
命令行太长。
CC out\py_engine\external\app_mgr\app_message.o
命令行太长。
CC out\py_engine\external\app_mgr\app_mgr.o
命令行太长。
CC out\py_engine\external\app_mgr\app_upgrade.o
命令行太长。
CC out\py_engine\external\unzip\src\adler32.o
scons: *** [out\py_engine\adapter\esp32\main.o] Error 1
scons: *** [out\py_engine\external\amp_task\amp_task.o] Error 1
scons: *** [out\py_engine\external\amp_boot\amp_boot.o] Error 1
scons: *** [out\py_engine\external\amp_boot\amp_boot_cmd.o] Error 1
scons: *** [out\py_engine\external\amp_boot\amp_boot_file_transfer.o] Error 1
scons: *** [out\py_engine\external\amp_boot\amp_boot_uart.o] Error 1
scons: *** [out\py_engine\external\amp_boot\amp_boot_ymodem.o] Error 1
scons: *** [out\py_engine\external\app_mgr\app_message.o] Error 1
命令行太长。
scons: *** [out\py_engine\external\app_mgr\app_mgr.o] Error 1
命令行太长。
scons: *** [out\py_engine\external\app_mgr\app_upgrade.o] Error 1
命令行太长。
scons: *** [out\py_engine\external\unzip\src\adler32.o] Error 1
scons: building terminated because of errors.
make: *** [all] Error 2
Build takes 0 Minutes 17 Seconds.

skylarCai, 2022-02-25:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

junchengliu, 2022-02-28:
@rualdeng this issues has solved, please udpate the last version code and compile again, If you have any questions, please let me know in time, thanks

closed 2022-03-07 by skylarCai