alibaba/AliOS-Things #1378: 温度计实验一直报发送失败

wenkaitiantian, 2022-02-28:
温度计实验一直报发送失败
[image]

skylarCai, 2022-02-28:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zy20220224, 2022-02-28:
> 温度计实验一直报发送失败 [image]

你用的是哪个案例？

wenkaitiantian, 2022-02-28:
https://gitee.com/haasedu/haasedu/tree/master/1-%E5%BF%AB%E9%80%9F%E5%BC%80%E5%A7%8B/2-%E4%BC%A0%E6%84%9F%E5%99%A8/%E4%BB%A3%E7%A0%81/HaaS_EDU_K1/Python

ethanlcz, 2022-03-07:
该问题2月28日已经fix。
[image]

closed 2022-03-07 by wenkaitiantian