alibaba/AliOS-Things #1379: 插入以太网

uofjava, 2022-02-28:
分支：master
案例名称:holleword
运行环境:c语言
固件版本:minilibc:  V7.2.0   aos: >=7.2.0 
关键代码:开发板插入以太网（可上网）以太网闪烁10次左右，重启
运行日志:

skylarCai, 2022-02-28:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

qisheng-seu, 2022-02-28:
亲，请选择linksdk_demo@haas100案例（即“Wi-Fi设备连接阿里云示例”）编译的固件才支持以太网驱动哦，关于下载案例源码，编译，烧录请参考下面文档。
https://haas.iot.aliyun.com/aliosthings/linksdk_demo.html

uofjava, 2022-02-28:
在依赖里面添加linksdk，不行吗？

# depends:                                 # <可选项> 该组件依赖其他的组件，合理的依赖才能保证组件能编译、使用
#   - minilibc: 'v7.2.0'
#   - aos: 'v7.2.0 ? <ENABLE_AOS>'
depends:
  - init: master
  - cli: master
  - osal_aos: master
  - mbmaster: master
  - linksdk: master
  - mbedtls: master
  - netmgr: master
  - lwip: master
  - uservice: master
  - wifi: master
  - haas100:  master
supported_boards:
  - haas100: master
  - haas200: master
  - haaseduk1: master

qisheng-seu, 2022-02-28:
如AliOS Things 3.3的rel_3.3.0分支中的源码所示，linksdk_demo中的package.yaml中除了依赖linksdk组件之外，还有几个.c源码需要添加上，否则linksdk并不会被初始化调用。为了降低开发者使用linksdk通过wifi或者以太网上云的门槛，我们特意设计了linksdk_demo给大家快速上手使用。因此建议您选择rel_3.3.0分支的solution/linksdk_demo案例来编译使用，谢谢。
[image]

uofjava, 2022-02-28:
好的

closed 2022-03-07 by skylarCai