alibaba/AliOS-Things #1380: haas100wifi连接错误

uofjava, 2022-03-01:
wifi突然不能使用（通过手机连接wifi是可以正常使用的），日志（重连）：
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
WIFI_DEV_CMD_CANCEL_CONNECT failed
    44710/reconnect | bwifi_disconnect state:CONNECTED
    44710/reconnect | [WIFI] change status:CONNECTED->DISCONNECTING
    44710/reconnect | wpa_supplicant_deauthenticate 744 0xc0b3b4f.
    44710/reconnect | ieee80211_set_disassoc 0 1

    44711/reconnect | cw1200_bss_info_changed=0x2000 hw_priv:0x20038fd8 priv:0x20039d20  

    44711/reconnect | [STA] Freq 2442 (wsm ch: 7).

    44711/reconnect | cw1200_bss_info_changed=0x9f hw_priv:0x20038fd8 priv:0x20039d20    

    44712/reconnect | BSS_CHANGED_ASSOC.

    44712/reconnect | [STA] Slot time :20 us.

    44712/reconnect | [CQM] RSSI threshold subscribe: 0 +- 0

    44712/reconnect | [CQM] Beacon loss subscribe: 0

    44712/reconnect | [CQM] TX failure subscribe: 0

    44712/reconnect | wlan0: deauthenticating from 0x3404a710M by local choice (reason=3)

    44712/cw1200_bh | [WSM] Issue unjoin command (TX).

    44712/reconnect | nl80211_send_deauth

    44717/net_wq | [STA] Unjoin.

    44717/reconnect | bwifi send event, id:3
    44718/reconnect | cw1200_bss_info_changed=0x4000 hw_priv:0x20038fd8 priv:0x20039d20

    44718/reconnect | ignore IEEE80211_CONF_CHANGE_MONITOR (0) IEEE80211_CONF_CHANGE_IDLE (1)

    44718/wifi_even | [WIFI] status DISCONNECTING, event 3
    44718/wifi_even | disconnected from AP
    44718/wifi_even | on_wifi_disconnect event_id:3
    44718/wifi_even | wifi disconnected (reason=3, locally_generated=1)
    44718/wifi_even | [WIFI] change status:DISCONNECTING->IDLE
wifi_connect_failed_cb:300 err=4124
    44719/reconnect | disconnect complete, success
get_wifi_config:1111 open /data/temp/temp.conf failed:No such file or directory
ssid_contain_chinese_check:2191 Error: app ssid is NULL!
    44723/reconnect | Scan for specific SSID TP-workS
    44723/reconnect | [WIFI] change status:IDLE->SCANNING
    44723/reconnect | nl80211: request->n_ssids=1, request->n_channels=13
    44723/reconnect | cw1200_bss_info_changed=0x4000 hw_priv:0x20038fd8 priv:0x20039d20

    44723/reconnect | ignore IEEE80211_CONF_CHANGE_MONITOR (0) IEEE80211_CONF_CHANGE_IDLE (1)

    44723/reconnect | [SCAN] Scan request for 1 SSIDs.

    44762/net_taskl | ieee80211_scan_rx: find qualified SSID in ProbeResp (2442,"TP-workS",-53,"74:05:a5:7a:5b:67")
    44762/net_taskl | [SCAN] Cancel scan in progress

    44764/cw1200_bh | cw1200_scan_complete_cb status:0
    44764/net_wq | [SCAN] Scan canceled.

    44764/net_wq | cw1200_bss_info_changed=0x4000 hw_priv:0x20038fd8 priv:0x20039d20

    44764/net_wq | ignore IEEE80211_CONF_CHANGE_MONITOR (0) IEEE80211_CONF_CHANGE_IDLE (1)

    44764/net_wq | scan result0:(2442,"TP-workS",-53,"74:05:a5:7a:5b:67")
    44764/net_wq | bwifi send event, id:0
    44765/wifi_even | [WIFI] status SCANNING, event 0
    44765/wifi_even | scan complete, status:1
    44765/reconnect | [WIFI] change status:SCANNING->IDLE
[... 128 lines cut ...]
    44955/net_taskl | eapol_sm_step 920 0xc0c53a9.
    44955/net_taskl | eap_peer_sm_step 1485.
    44955/net_taskl | eapol_sm_step 920 0xc0c53df.
    44955/net_taskl | EAPOL: SUPP_PAE entering state AUTHENTICATING
    44955/net_taskl | EAPOL: SUPP_BE entering state SUCCESS
    44955/net_taskl | eap_peer_sm_step 1485.
    44955/net_taskl | EAP: EAP entering state DISABLED
    44955/net_taskl | eap_peer_sm_step 1485.
    44955/net_taskl | EAPOL: SUPP_PAE entering state AUTHENTICATED
    44955/net_taskl | EAPOL: SUPP_BE entering state IDLE
    44955/net_taskl | eap_peer_sm_step 1485.
    44955/net_taskl | eap_peer_sm_step 1485.
    44955/wifi_even | [WIFI] status CONNECTING, event 1
    44955/wifi_even | wifi connecting in state 9
    44955/wifi_even | on_wifi_connect event_id:0
    44955/wifi_even | [WIFI] status CONNECTING, event 2
    44955/wifi_even | wifi mac connected to TP-workS
    44955/wifi_even | on_wifi_connect event_id:1
    44956/wifi_even | [WIFI] change status:CONNECTING->CONNECTED
    44956/wifi_even | wsm_epta_cmd 60000 40000 3.
    44956/wifi_even | Set PS mode 0
    44956/wifi_even | nl80211_set_power_save disabled
    44956/wifi_even | config_dpd, ch=7
    44957/wifi_conn | connect complete, success
ssid_contain_chinese_check:2191 Error: app ssid is NULL!

skylarCai, 2022-03-01:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

uofjava, 2022-03-01:
人工支持

uofjava, 2022-03-01:
拔电重新启动，又可以了

YiluMao, 2022-03-01:
好的，有问题请继续提问

qisheng-seu, 2022-03-01:
44955/wifi_even | wifi mac connected to TP-workS
44956/wifi_even | [WIFI] change status:CONNECTING->CONNECTED
44957/wifi_conn | connect complete, success
日志显示wifi mac连接已经成功了。

ssid_contain_chinese_check:2191 Error: app ssid is NULL! 这个日志打印不影响的，这是检查ssid是否包含中文字符，目前是不包含中文字符的。但是我们的代码打印  app ssid is NULL  确实不太规范。我们记录下来改进。

谢谢！！

closed 2022-03-07 by skylarCai