alibaba/AliOS-Things #1384: [Feature Request]: how to operate w25xx with driver/SPI

sopadilla, 2022-03-04:
### Contact Details

liangyinhao44@163.com

### What do you want?

how can I operate w25qxx with driver/SPI interfaces in haas2.0?
i found that when i try to communicate to w25qxx with existing apis，it doesn't work。

### Version

master (Default)

skylarCai, 2022-03-04:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

wuxn9999, 2022-03-04:
请问您使用的是哪一种硬件？
使用的是C语言SPI API还是Python轻应用的SPI API？

sopadilla, 2022-03-07:
我用的是python轻应用的haas2.0的api接口

wuxn9999, 2022-03-16:
我们这里有SPI的Python API文档。[https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/haas_extended_api/driver/spi](url)
您可以对照w25qxx的手册，通过SPI API来操作flash。

wuxn9999, 2022-03-16:
我们这里有SPI的[Python API文档](https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/haas_extended_api/driver/spi)。
您可以对照w25qxx的手册，通过SPI API来操作flash。

wuxn9999, 2022-03-16:
另外您用的是哪一款开发板？您现在操作w25qxx的代码方便发一下吗？

YiluMao, 2022-03-23:
@sopadilla 请问您的问题有没有解决，如果有解决，请关闭issue，多谢

closed 2022-03-24 by skylarCai