alibaba/AliOS-Things #1386: [Bug Report]: 播放音频代码报错

AnsonCode, 2022-03-05:
### What happened?

A bug happened!
运行环境: 从官网下载的[固件](https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16834026.0.0.4e1260b1OEE04B#/JavaScript/docs/zh-CN/startup/haaseduk1-quick-start?id=_2%e3%80%81js%e8%bd%bb%e5%ba%94%e7%94%a8%e5%9b%ba%e4%bb%b6)
固件版本: haas edu k1

按照[音频播放示例](https://help.aliyun.com/document_detail/268727.html?spm=a2c4g.11186623.6.672.6f6050c2lTw4vV)发生如下报错。

```sh
begin eval file: /data/jsamp/app.js
begin load module audioplayer
find binary module audioplayer, len 1317, begin load
begin load module events
find binary module events, len 5023, begin load
load module events success
load module audioplayer success
js engine eval /data/jsamp/app.js error
app.js content: 
import * as audioplayer from 'audioplayer';
var audioplayerState = ['stop', 'paused', 'playing', 'listplay_begin', 'listplay_end', 'error'];
audioplayer.on('stateChange', function (state) {
  console.log('audioplayer state: ' + audioplayerState[state]);
});
audioplayer.setVolume(6);
var audio = '01start.mp3';
audioplayer.play(audio, function () {
  console.log('playback complete');
});
setTimeout(function () {
  console.log("playback pause");
  audioplayer.pause();
  var position = audioplayer.getPosition();
  var duration = audioplayer.getDuration();
  console.log('playback progress: ' + position + '/' + duration);
}, 5000);
setTimeout(function () {
  console.log("playback resume");
  audioplayer.resume();
}, 10000);
setTimeout(function () {
  console.log("playback from 1s");
  audioplayer.seekto(1);
}, 15000);
setTimeout(function () {
  var position = audioplayer.getPosition();
  var duration = audioplayer.getDuration();
  console.log('playback progress: ' + position + '/' + duration);
}, 16000);
setTimeout(function () {
  console.log("playback stop");
  audioplayer.stop();
}, 30000);
var app = {};
export default app;
TypeError: not a function    at <anonymous> (/data/jsamp/app.js:5)
```

[image]
[image]

### Version

master (Default)

### What soultions are you seeing the problem on?

audio_demo

skylarCai, 2022-03-05:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

qisheng-seu, 2022-03-07:
亲，您参考的 [音频播放](https://help.aliyun.com/document_detail/268727.html)示例是HaaS600上的示例，HaaS EDU K1上暂时不支持该案例。HaaS EDU K1上的智能语音方案可以参考[HaaS云端一体智能语音交互方案](https://blog.csdn.net/HaaSTech/article/details/121018254?ops_request_misc=%257B%2522request%255Fid%2522%253A%2522164661888516781683917906%2522%252C%2522scm%2522%253A%252220140713.130102334.pc%255Fblog.%2522%257D&request_id=164661888516781683917906&biz_id=0&utm_medium=distribute.pc_search_result.none-task-blog-2~blog~first_rank_ecpm_v1~rank_v31_ecpm-17-121018254.nonecase&utm_term=%E9%9F%B3%E9%A2%91&spm=1018.2226.3001.4450)

AnsonCode, 2022-03-07:
可否提供下JS的示例代码呢？

qisheng-seu, 2022-03-07:
JS示例依赖于芯片的固件是否支持。目前已经发布的haaseduk1 js固件还不支持语音播放的功能。

closed 2022-03-16 by skylarCai