alibaba/AliOS-Things #1403: 运行Demo报错

bawkss, 2022-03-06:
运行HTTP Demo后报错没有输出，不知道哪里错了，板子HaasEC600s
app.js代码
`var httpc = require('http');
var network = require('network');
var net = network.openNetWorkClient();

var request_url = 'http://appengine.oss-cn-hangzhou.aliyuncs.com/httpTest.txt';

function http_get() {
  httpc.request({
    url: request_url,
    method: 'GET',
    timeout: 1000,
    headers: {
      'content-type': 'application/json',
    },
    success: function (data) {
      console.log('http: [debug] receive data is ' + data);
    }
  });
}

function http_post() {
  httpc.request({
    url: request_url,
    method: 'POST',
    timeout: 1000,
    headers: {
      'content-type': 'application/json',
    },
    success: function (data) {
      console.log('http: [debug] receive data is ' + data);
    }
  });
}

var netStu = net.getStatus();

if (netStu == 'connect') {
  http_get();
} else {
  net.on('connect', function() {
    http_get();
  })
}`

app.json代码
`{
    "version": "1.0.0",
    "io": {},
    "debugLevel": "DEBUG"
  }`

烧录
`Windows PowerShell
版权所有（C） Microsoft Corporation。保留所有权利。

安装最新的 PowerShell，了解新功能和改进！https://aka.ms/PSWindows

PS C:\Users\chenbaolin\workspace> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path" ; echo "=> powershell.exe detected, set miniconda env finished."
=> powershell.exe detected, set miniconda env finished.
PS C:\Users\chenbaolin\workspace> c:\Users\chenbaolin\.vscode\extensions\haas.haas-studio-1.0.68/amp/amp-win/dist/amp.exe serialput c:\Users\chenbaolin\workspace/solutions/ COM5                 -m quickjs -e 0 -b 115200 -p c:\Users\chenbaolin\.vscode\extensions\haas.haas-studio-1.0.68/amp/amp-win
device type: ymodem
AMP FILE NOT EXIST
PS C:\Users\chenbaolin\workspace> c:\Users\chenbaolin\.vscode\extensions\haas.haas-studio-1.0.68/amp/amp-win/dist/amp.exe serialput c:\Users\chenbaolin\workspace/solutions/ COM5                 -m quickjs -e 0 -b 115200 -p c:\Users\chenbaolin\.vscode\extensions\haas.haas-studio-1.0.68/amp/amp-win
device type: ymodem
AMP FILE NOT EXIST
PS C:\Users\chenbaolin\workspace> c:\Users\chenbaolin\.vscode\extensions\haas.haas-studio-1.0.68/amp/amp-win/dist/amp.exe serialput c:\Users\chenbaolin\workspace/solutions/test_demo_http COM5                 -m quickjs -e 0 -b 115200 -p c:\Users\chenbaolin\.vscode\extensions\haas.haas-studio-1.0.68/amp/amp-win
device type: ymodem
generating app package...
[... 369 lines cut ...]
    at [anon] (amp/engine/duktape_engine/duktape/duktape.c:77198) internal

AT command server ready!
+CEREG: 0
+CEREG: 0
Booting done
5
PMIC_Phase3Init,Clk Init
send 0xee to DSP 
SW restart reason: Normal powering on
CommPMLPT --- 1, 0, 1
+CEREG: 1,"702b","04a6c884",7
[PMIC_RTC_SetTime] type = SYS_OFFSET, m_rtc_app_sys_keeping_synced = 1 
[PMIC_RTC_SetTime] [6224f0e4][20c][6224eed8]:2022-03-06 17:35:32 [SUN]
[PMIC_RTC_Alarm_Update_Follow_RTC_SetTime] src_offset=[0x6224eed8] dst_offset=[0x6224eed8], EXPIRE_ID=[1]
[pmic_rtc] sys_offset=[0x6224eed8],app_offset=[0x6224eed8],timezone=[8]
[pmic_rtc] sys_offset=[0x6224eed8],app_offset=[0x6224eed8],timezone=[8]
[pmic_rtc] sys_offset=[0x6224eed8],app_offset=[0x6224eed8],timezone=[8]
PMIC_RTC_SetTimezone:206,GLOBAL_TIMEZONE=[8]
[PMIC_RTC_USERDATA_CFG_GET] bit_mask=[2]
[pmic_userdata_get_app_rtc_ever_configed]=[0]
[PMIC_RTC_GetTime_Count] SYS_OFFSET rtc_cnt   = 20d
[PMIC_RTC_GetTime_Count] SYS_OFFSET rtc_offset = 6224eed8
get_operator_name_task
`

skylarCai, 2022-03-06:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

bawkss, 2022-03-06:
人工支持

bawkss, 2022-03-06:
已解决，引擎选择错误
在`.vscode\upgrade.json`下配置  `"engine": "quickjs"`   改为  `"engine": "duktape"`
可以关闭Issue了，谢谢。

closed 2022-03-06 by bawkss