alibaba/AliOS-Things #1406: haas100轻应用固件编译无法通过

xyjand, 2022-03-07:
haas100，烧录轻应用固件：
1）我的环境：win11+vscode最新版+haas studio(v1.0.70) C语言+轻应用示例  ；编译一直出问题，无法继续；
2）为此在虚机里另安装了centos，win7 64系统，vscode 同样在编译出问题，无法通过。

skylarCai, 2022-03-07:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

qisheng-seu, 2022-03-07:
亲，辛苦重新创建个"C语言+轻应用示例"的源码工程，10分钟前我们刚解决了编译错误的问题，谢谢

qisheng-seu, 2022-03-07:
亲，amp_demo@haas100的编译问题我们10分钟前已经修复。请基于haas stduio重新下载个源码工程编译。谢谢。

xyjand, 2022-03-07:
haas-studio更新到了最新版 V1.0.71 ;  重新下载了 轻应用示例，编译时还是有错误。

CC out\amp_adapter\platform\aos\aos_fs.o
CC out\amp_adapter\platform\aos\haas100\amp_ota_port.o
CC out\amp_adapter\platform\aos\aos_system.o
CC out\amp_adapter\platform\aos\network\aos_cellular.o
CC out\amp_adapter\platform\aos\network\aos_httpc.o
CC out\amp_adapter\platform\aos\network\aos_netdev.o
CC out\amp_adapter\platform\aos\network\aos_tcp.o
CC out\amp_adapter\platform\aos\network\aos_udp.o
CC out\amp_adapter\platform\aos\network\aos_wifi.o
CC out\amp_adapter\platform\aos\peripheral\aos_hal_adc.o
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\aos_system.c: In function 'aos_network_status_registercb':
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\aos_system.c:117:12: warning: implicit declaration of function 'aos_wifi_set_msg_cb'; did you mean 'netmgr_wifi_set_msg_cb'? [-Wimplicit-function-declaration]
  117 |     return aos_wifi_set_msg_cb(cb);
      |            ^~~~~~~~~~~~~~~~~~~
      |            netmgr_wifi_set_msg_cb
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\aos_system.c: In function 'aos_get_network_status':
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\aos_system.c:124:11: warning: implicit declaration of function 'aos_wifi_get_state'; did you mean 'netmgr_wifi_get_state'? [-Wimplicit-function-declaration]
  124 |     ret = aos_wifi_get_state();
      |           ^~~~~~~~~~~~~~~~~~
      |           netmgr_wifi_get_state
CC out\amp_adapter\platform\aos\peripheral\aos_hal_can.o
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\aos_system.c: In function 'aos_get_device_name':
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\aos_system.c:173:5: warning: implicit declaration of function 'tg_get_chipid' [-Wimplicit-function-declaration]
  173 |     tg_get_chipid(chip_id, 16);
      |     ^~~~~~~~~~~~~
CC out\amp_adapter\platform\aos\peripheral\aos_hal_flash.o
CC out\amp_adapter\platform\aos\peripheral\aos_hal_gpio.o
AR aos_sdk\lib\libkv.a
CC out\amp_adapter\platform\aos\peripheral\aos_hal_i2c.o
RANLIB aos_sdk\lib\libkv.a
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\network\aos_tcp.c: In function 'aos_tcp_establish':
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\network\aos_tcp.c:83:9: warning: implicit declaration of function 'close'; did you mean 'pclose'? [-Wimplicit-function-declaration]
   83 |         close(fd);
      |         ^~~~~
      |         pclose
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\network\aos_tcp.c: In function 'aos_tcp_read':
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\network\aos_tcp.c:204:5: CC out\amp_adapter\platform\aos\peripheral\aos_hal_lcd.o
warning: implicit declaration of function 'gettimeofday' [-Wimplicit-function-declaration]
  204 |     gettimeofday(&timestart, NULL);
      |     ^~~~~~~~~~~~
CC out\amp_adapter\platform\aos\peripheral\aos_hal_pwm.o
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\network\aos_udp.c: In function 'aos_udp_socket_bind':
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\network\aos_udp.c:56:9: warning: implicit declaration of function 'close'; did you mean 'pclose'? [-Wimplicit-function-declaration]
   56 |         close(p_socket);
      |         ^~~~~
      |         pclose
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\network\aos_udp.c: In function 'aos_udp_read_timeout':
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\network\aos_udp.c:358:13: warning: implicit declaration of function 'read'; did you mean 'fread'? [-Wimplicit-function-declaration]
  358 |     return  read((long)p_socket, p_data, datalen);
      |             ^~~~
      |             fread
CC out\amp_adapter\platform\aos\peripheral\aos_hal_rtc.o
CC out\amp_adapter\platform\aos\peripheral\aos_hal_spi.o
CC out\amp_adapter\platform\aos\peripheral\aos_hal_timer.o
CC out\amp_adapter\platform\aos\peripheral\aos_hal_uart.o
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\peripheral\aos_hal_gpio.c: In function 'aos_hal_gpio_init':
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\peripheral\aos_hal_gpio.c:129:23: warning: passing argument 1 of 'aos_gpioc_put' from incompatible pointer type [-Wincompatible-pointer-types]
  129 |         aos_gpioc_put(&gpioc);
      |                       ^~~~~~
      |                       |
      |                       aos_gpioc_ref_t ** {aka struct aos_dev_ref **}
In file included from E:\PROJECT\haas\h4\components\amp_adapter\include\peripheral/aos_hal_gpio_internal.h:21,
                 from E:\PROJECT\haas\h4\components\amp_adapter\include\peripheral/aos_hal_gpio.h:4,
                 from E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\peripheral\aos_hal_gpio.c:12:
E:\PROJECT\haas\h4\components\drivers\peripheral\gpio\include/aos/gpioc.h:36:37: note: expected 'aos_gpioc_ref_t *' {aka 'struct aos_dev_ref *'} but argument is of type 'aos_gpioc_ref_t **' {aka 'struct aos_dev_ref **'}
   36 | void aos_gpioc_put(aos_gpioc_ref_t *ref);
      |                    ~~~~~~~~~~~~~~~~~^~~
E:\PROJECT\haas\h4\components\amp_adapter\platform\aos\peripheral\aos_hal_lcd.c: In function 'aos_hal_lcd_uninit':
[... 855 lines cut ...]
E:\PROJECT\haas\h4\hardware\board\haas100\config/board.h:5: warning: "DEBUG_LASTWORD_RAM_ADDR" redefined
    5 | #define DEBUG_LASTWORD_RAM_ADDR 0x20166000
      |
<command-line>: note: this is the location of the previous definition
CC out\ota\hal\ota_hal_vfs_param.o
scons: *** [aos_sdk\lib\libgpio.a] Error 1
[... cut ...]
      |                                             int (*)(void)
In file included from amp_entry.c:8:
E:\PROJECT\haas\h4\components\osal_aos\include/aos/kernel.h:1113:74: note: expected 'void (*)(void *)' but argument is of type 'int (*)(void)'
 1113 | aos_status_t aos_task_new_ext(aos_task_t *task, const char *name, void (*fn)(void *),
      |                                                                   ~~~~~~~^~~~~~~~~~~
scons: building terminated because of errors.
make: *** [all] Error 2
Build takes 1 Minutes 1 Seconds.

终端将被任务重用，按任意键关闭。

closed 2022-03-16 by skylarCai