alibaba/AliOS-Things #1413: haas100开发板插入网线网络不稳定

uofjava, 2022-03-11:
在开发板中写入mqttdemo(连入局域网mqtt服务器)发布消息，不定时掉网（处理：重新开启mqtt）日志：
MQTTPublish 0, msgid 1281
MQTTPublish 0, msgid 1282
MQTTPublish 0, msgid 1283
MQTTPublish 0, msgid 1284
MQTTPublish 0, msgid 1285
MQTTPublish 0, msgid 1286
MQTTPublish 0, msgid 1287
MQTTPublish -1, msgid 1288
[1716.657]<I>DEBUG [ LwIP ] LwIP_send, pkt:0x2005960c, netif(0x3468757c), 192.168.3.148 -> 192.168.3.113 IPID(1097), S:7d7e8, A:28435f29, W:14280, l:0, 52544 -> 1883, TCP_FINACK(40)
[1716.659]<I>DEBUG [ LwIP ] LwIP_send, pkt:0x200597f8, netif(0x3468757c), 192.168.3.148 -> 192.168.3.113 IPID(1098), S:7eb43, A:0, W:14600, l:0, 52545 -> 1883, TCP_SYN(44)
[1716.666]<I>DEBUG [ LwIP ] LwIP_send, pkt:0x200597f8, netif(0x3468757c), 192.168.3.148 -> 192.168.3.113 IPID(1099), S:7eb43, A:0, W:14600, l:0, 52545 -> 1883, TCP_SYN(44)
[1716.667]<I>DEBUG [ LwIP ] LwIP_send, pkt:0x2005960c, netif(0x3468757c), 192.168.3.148 -> 192.168.3.113 IPID(109b), S:7d7e8, A:28435f29, W:14280, l:0, 52544 -> 1883, TCP_FINACK(40)
[1716.682]<I>DEBUG [ LwIP ] LwIP_recv, pkt:0x2005a0d8, netif(0x3468757c), 192.168.3.113 -> 192.168.3.148 IPID(3d1f), S:28435f2d, A:7d7e8, W:29200, l:0, 1883 -> 52544, TCP_FINACK(40)
[1716.687]<I>DEBUG [ LwIP ] LwIP_recv, pkt:0x2005a6f8, netif(0x3468757c), 192.168.3.113 -> 192.168.3.148 IPID(3d20), S:28435f29, A:7d7e8, W:29200, l:4, 1883 -> 52544, TCP_FINPSHACK(44)
[1716.688]<I>DEBUG [ LwIP ] LwIP_send, pkt:0x2005958c, netif(0x3468757c), 192.168.3.148 -> 192.168.3.113 IPID(109e), S:7d7e9, A:28435f2e, W:14600, l:0, 52544 -> 1883, TCP_RSTACK(40)
[1716.777]<I>DEBUG [ LwIP ] LwIP_recv, pkt:0x2005a0d8, netif(0x3468757c), 192.168.3.113 -> 192.168.3.148 IPID(0), S:54a1638e, A:7eb44, W:29200, l:0, 1883 -> 52545, TCP_SYNACK(44)
Network Connect Success!
Connecting to 192.168.3.113:1883
[1716.782]<I>DEBUG [ LwIP ] LwIP_send, pkt:0x20059628, netif(0x3468757c), 192.168.3.148 -> 192.168.3.113 IPID(10a1), S:7d7e8, A:28435f2e, W:14600, l:0, 52544 -> 1883, TCP_RSTACK(40)
[1716.787]<I>DEBUG [ LwIP ] LwIP_send, pkt:0x20059628, netif(0x3468757c), 192.168.3.148 -> 192.168.3.113 IPID(10a2), S:7d7e9, A:28435f2e, W:14600, l:0, 52544 -> 1883, TCP_RSTACK(40)
MQTT Connect Success!
MQTTPublish 0, msgid 1289
MQTTPublish 0, msgid 1290
MQTTPublish 0, msgid 1291
MQTTPublish 0, msgid 1292
MQTTPublish 0, msgid 1293
MQTTPublish 0, msgid 1294
MQTTPublish 0, msgid 1295
MQTTPublish 0, msgid 1296
MQTTPublish 0, msgid 1297
MQTTPublish 0, msgid 1298

skylarCai, 2022-03-11:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

qisheng-seu, 2022-03-11:
亲，请帮忙告知你用什么开发语言，使用什么示例代码测试的结果呢？

6636549, 2022-03-12:
在开发板中写入mqttdemo(连入局域网mqtt服务器)发布消息，不定时掉网（处理：重新开启mqtt）
日志：
重启也没办法啊。我这边。大概5分钟就需要重启一次，每次重启，数据丢书超过1秒以上

uofjava, 2022-03-12:
> 亲，请帮忙告知你用什么开发语言，使用什么示例代码测试的结果呢？

使用C语言，在helloworld中引入mqtt列子，并加入网络连接事件监听，开启连接mqtt任务

uofjava, 2022-03-12:
入口代码：/*
 * Copyright (C) 2015-2020 Alibaba Group Holding Limited
 */

#include "aos/init.h"
#include "board.h"
#include <aos/errno.h>
#include <aos/kernel.h>
#include <k_api.h>
#include <stdio.h>
#include <stdlib.h>
#include "netmgr.h"
#include <uservice/uservice.h>
#include <uservice/eventid.h>

extern int mqtt_main(int argc, char *argv[]);

static int _ip_got_finished = 0;

static void entry_func(void *data)
{
    printf("start mqtt main");
    mqtt_main(0 , NULL);

}

static void wifi_event_cb(uint32_t event_id, const void *param, void *context)
{
    aos_task_t task;
    aos_status_t ret;
    if (event_id != EVENT_NETMGR_DHCP_SUCCESS)
        return;

    if (_ip_got_finished != 0)
        return;

    _ip_got_finished = 1;
    ret = aos_task_create(&task, "mqtt_demo", entry_func,
                          NULL, NULL, 6048, AOS_DEFAULT_APP_PRI, AOS_TASK_AUTORUN);
    if (ret < 0) {
        printf("create linksdk demo task failed, ret = %ld \r\n", ret);
    }
}

int application_start(int argc, char *argv[])
{
    int count = 0;

    printf("nano entry here!\r\n");

    aos_set_log_level(AOS_LL_DEBUG);

    
    event_service_init(NULL);

    netmgr_service_init(NULL);
    // netmgr_set_auto_reconnect(NULL, true);
    // netmgr_wifi_set_auto_save_ap(true);

    event_subscribe(EVENT_NETMGR_DHCP_SUCCESS, wifi_event_cb, NULL);
    while (1) {
        aos_msleep(1000);

    };
}

uofjava, 2022-03-12:
mqtt业务代码：
/**
 * @copyright Copyright (C) 2015-2021 Alibaba Group Holding Limited
 */

#include <stdio.h>
#include <memory.h>
#include "MQTTClient.h"

#include <stdio.h>
#include <signal.h>
#include <sys/time.h>
#include "led.h"

Network n;
MQTTClient c;
unsigned char buf[2000];
unsigned char readbuf[2000];
char *host = "192.168.3.113";  //使用mosquitto搭建的linux服务器
short port = 1883;
const char *subTopic = "test/aiot/sub_topic";
const char *pubTopic = "test";
char clientId[150] = "haas";
char username[65] = {"yyt"};
char password[65] = {"yyt"};
int rc = 0;
int netInit()
{
	NetworkInit(&n);
	rc = NetworkConnect(&n, host, port);
	if(rc != 0){
		printf("Network Connect failed:%d\n", rc);
		return -1;
	}else{
		printf("Network Connect Success!\n");
	}
	return 0;
}
int initMMQTT()
{
	/* init mqtt client */
	MQTTClientInit(&c, &n, 1000, buf, sizeof(buf), readbuf, sizeof(readbuf));
	/* set mqtt connect parameter */
	MQTTPacket_connectData data = MQTTPacket_connectData_initializer;
	data.willFlag = 0;
	data.MQTTVersion = 3;
	data.clientID.cstring = clientId;
	data.username.cstring = username;
	data.password.cstring = password;
	data.keepAliveInterval = 60;
	data.cleansession = 1;
	printf("Connecting to %s:%d\n", host, port);
	rc = MQTTConnect(&c, &data);
	if(rc != 0){
		printf("MQTT Connect server failed:%d\n", rc);
		return;
	}else{
		printf("MQTT Connect Success!\n");
	}
	return 0;
}

int mqtt_main(int argc, char** argv)
{
	int back = netInit();
	while (back == -1)
	{
		led_switch(2, LED_ON);
		/* code */
		back = netInit();
		aos_msleep(1000);
	}
	led_switch(2, LED_OFF);
	initMMQTT();
	int cnt = 0;
    unsigned int msgid = 0;
	while (1)
	{
		// MQTTYield(&c, 5000);
		aos_msleep(300);
		char *send_alio_data = (char *)calloc(100,sizeof(char));
        strcat(send_alio_data,"{");
		char d [30];
		sprintf(d,"\"bar\": %d",msgid);
		strcat(send_alio_data,d);
		strcat(send_alio_data, "}");
		MQTTMessage msg = {
			QOS1,
			0,
			0,
			0,
			send_alio_data,
			strlen(send_alio_data),
		};
		msg.id = ++msgid;
		// printf(send_alio_data);
		rc = MQTTPublish(&c, pubTopic, &msg);
		printf("MQTTPublish %d, msgid %d\n", rc, msgid);
		aos_free(send_alio_data);
		if(rc ==-1){
			MQTTDisconnect(&c);
			NetworkDisconnect(&n);
			back = netInit();
			while (back == -1)
			{
				led_switch(2, LED_ON);
				/* code */
				back = netInit();
				aos_msleep(1000);
			}
			led_switch(2, LED_OFF);
			initMMQTT();
		}
	}
	printf("MQTT example exit!\n");
	MQTTDisconnect(&c);
	NetworkDisconnect(&n);
	return 0;
}

yanxiaoyong-ch, 2022-03-16:
从开机的到出问题的完整日志麻烦贴出来一下

YiluMao, 2022-03-23:
@uofjava 
请帮忙提供下完整日志

closed 2022-03-24 by skylarCai