alibaba/AliOS-Things #1415: [Feature Request]: inksdk_demo 编译报错

frankshen2020, 2022-03-13:
### Contact Details

shenchendky@163.com

### What do you want?

collect2.exe: error: ld returned 1 exit status
scons: *** [out\test_demo\aos.elf] Error 1
scons: building terminated because of errors.
make: *** [all] Error 2
Build takes 0 Minutes 16 Seconds.

### Version

master (Default)

skylarCai, 2022-03-13:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

frankshen2020, 2022-03-13:
人工支持

YiluMao, 2022-03-13:
请问您如下是选择的什么？
[image]

YiluMao, 2022-03-15:
测试了HaaS100的硬件，wifi连接阿里云的解决方案
编译没有遇到问题

申请关闭本issue

closed 2022-03-15 by skylarCai