alibaba/AliOS-Things #1416: 有没有自定义Topic数据解析的案例

frankshen2020, 2022-03-13:
[自定义Topic数据解析](https://help.aliyun.com/document_detail/149962.html)
设备通过携带解析标记?_sn=default的自定义Topic上报数据 ， 而我使用的例程无法更改topic， 只能更改属性内容。  找不到topic在哪里定义的~

有没有自定义Topic数据解析的案例呢？

skylarCai, 2022-03-13:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-03-14:
> 有没有自定义Topic数据解析的案例呢？

阿里云物联网平台是支持自定义Topic，云端自定义Topic的操作步骤请参考：https://help.aliyun.com/document_detail/85539.htm?spm=a2c4g.11186623.0.0.41701d17ixgOSJ#concept-ppk-rz4-k2b
设备端通过HaaS Python aliyunIoT库对自定义Topic进行操作的API说明文档可以参考下面两个函数：
[subscribe - 订阅](https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16834026.0.0.595660b17TxBgM#/Python/docs/zh-CN/haas_extended_api/aliyunIoT/device?id=subscribe-%e8%ae%a2%e9%98%85)
[publish - 发布](https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16834026.0.0.595660b17TxBgM#/Python/docs/zh-CN/haas_extended_api/aliyunIoT/device?id=publish-%e5%8f%91%e5%b8%83)

frankshen2020, 2022-03-14:
可能我没表述清楚，我知道阿里云物联网平台可以自定义topic 。

 我是下位机想要进行透传发送数据，透传的数据在物联网平台解析。

 但是我没看到HAAS100 硬件的透传例程，我用的例程是“linksdk_demo Wi-Fi设备连接阿里云示例”，它里面的属性上报函数“demo_send_property_post”只能发物模型topic，也就是该topic不带“?_sn=default” 标记，那么这个消息就无法被识别进行解析，想问下有没有HASS100硬件，带“?_sn=default” 标记的，能够进行解析的示例

frankshen2020, 2022-03-14:
人工支持

YiluMao, 2022-03-23:
@frankshen2020 请问您的问题有没有被解决，如果已解决，请关闭issue

closed 2022-03-24 by skylarCai