alibaba/AliOS-Things #1417: [Bug Report]: 光敏电阻感应传感器和火焰传感器的错误

84788951, 2022-03-14:
### Contact Details

84788951@qq.com

### What happened?

A bug happened!
运行环境:esp32
固件版本:python 轻应用

光敏电阻感应传感器的
[image]
这里有问题。URL:https://haas.iot.aliyun.com/solution/detail/hardware?versionId=800C57784852F32C00000002&dataId=800C57784852F32C

火焰传感器
[image]
这里有问题。URL：https://haas.iot.aliyun.com/solution/detail/hardware?versionId=800C65AE63D505D100000002&dataId=800C65AE63D505D1
另外，火焰传感器的程序上传后，运行出错。
`=== upload progress end ===
3.1137757301330566
1.918652057647705
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
mount fs
 ==== python execute bootpy ====
enable OneMinuteOnCloud
 ==== python file check /data/pyamp/main.py ====
 ==== python execute from /data/pyamp/main.py ====
Testing photoresistor ...
Traceback (most recent call last):
  File "/data/pyamp/main.py", line 83, in <module>
AttributeError: 'module' object has no attribute 'Photoresistor'
mp_task-250 110712
MicroPython v1.17-dirty on 2022-01-20T11:58:13; ESP32 module with ESP32
Type "help()" for more information.
>>>`

`
import fire
from driver import ADC
print("Testing fire ...")
adcDev = ADC()
adcDev.open("gnss")
fireDev = fire.Fire(adcDev)    （第83行！！！！）
value = fireDev.getVoltage()
print("The fire status value ",value)
`

### Version

rel_3.3.0 (Main)

### What soultions are you seeing the problem on?

py_engine_esp32_demo

84788951, 2022-03-14:
import fire from driver 
import ADC print("Testing fire ...") 
adcDev = ADC() 
adcDev.open("gnss") 
fireDev = fire.Fire(adcDev) （第83行！！！！） 
value = fireDev.getVoltage() 
print("The fire status value ",value)

skylarCai, 2022-03-14:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zy20220224, 2022-03-14:
import fire from driver 这一行不对，应该是import fire

adcDev.open("gnss"), 这一行也不对，这里的“gnss”在board.json中没有配置

ethanlcz, 2022-03-14:
> import fire from driver

硬件积木说明中没有“import fire from driver”这句代码。
adcDev.open("gnss")这个已经修复

ethanlcz, 2022-03-14:
> `import fire from driver import ADC print("Testing fire ...") adcDev = ADC() adcDev.open("gnss") fireDev = fire.Fire(adcDev) （第83行！！！！） value = fireDev.getVoltage() print("The fire status value ",value)`

import fire和from driver import ADC之间是有分行的

ethanlcz, 2022-03-14:
> ### Contact Details
> [84788951@qq.com](mailto:84788951@qq.com)
> 
> ### What happened?
> A bug happened! 运行环境:esp32 固件版本:python 轻应用
> 
> 光敏电阻感应传感器的 [image] 这里有问题。URL:https://haas.iot.aliyun.com/solution/detail/hardware?versionId=800C57784852F32C00000002&dataId=800C57784852F32C
> 
> 火焰传感器 [image] 这里有问题。URL：https://haas.iot.aliyun.com/solution/detail/hardware?versionId=800C65AE63D505D100000002&dataId=800C65AE63D505D1 另外，火焰传感器的程序上传后，运行出错。 `=== upload progress end === 3.1137757301330566 1.918652057647705 --- Miniterm on COM3 115200,8,N,1 --- --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H --- mount fs ==== python execute bootpy ==== enable OneMinuteOnCloud ==== python file check /data/pyamp/main.py ==== ==== python execute from /data/pyamp/main.py ==== Testing photoresistor ... Traceback (most recent call last): File "/data/pyamp/main.py", line 83, in AttributeError: 'module' object has no attribute 'Photoresistor' mp_task-250 110712 MicroPython v1.17-dirty on 2022-01-20T11:58:13; ESP32 module with ESP32 Type "help()" for more information.
> 
> > > > `
> 
> `import fire from driver import ADC print("Testing fire ...") adcDev = ADC() adcDev.open("gnss") fireDev = fire.Fire(adcDev) （第83行！！！！） value = fireDev.getVoltage() print("The fire status value ",value)`
> 
> ### Version
> rel_3.3.0 (Main)
> 
> ### What soultions are you seeing the problem on?
> py_engine_esp32_demo
> 
> ### Relevant log output
> _No response_

此问题已经修复，请刷新火焰传感器页面查看。

84788951, 2022-03-14:
``@ethanlcz 我把修改过的页面上的例子跑了，还是出现错误。

`
[image]

`

ethanlcz, 2022-03-14:
> ``@ethanlcz 我把修改过的页面上的例子跑了，还是出现错误。
> 
> ` [image]
> 
> `
> 
> [image] 这里是否有错误。

抱歉，我手头没有开发板所以没有拿下来运行。我刚刚拿下来代码之后看有两个地方需要修改：
1. 将fire.Fire(adcDev)修改为：fire.FIRE(adcDev)
2. 将value = self.adcObj.read()修改为value = self.adcObj.readVoltage()

YiluMao, 2022-03-23:
@84788951 请问您的问题有没有被解决，如果已解决，请关闭issue

closed 2022-03-24 by skylarCai